innoveit/react-native-ble-manager #477: Set notification failed

marcoeg, 2019-02-21:
### Version

Tell us which versions you are using:

- react-native-ble-manager v6.6.2
- react-native v0.58.4
- iOS/Android v.7.0

### Expected behaviour

BleManager.startNotification() to return a Promise.

### Actual behaviour

Timeout with error:
"Set notification failed for 0000ff01-0000-1000-8000-00805f9b34fb"

### Steps to reproduce

Used basically the code in the example:

``
        BleManager.connect(peripheral.id).then(() => {
          let peripherals = this.state.peripherals;
          let p = peripherals.get(peripheral.id);
          if (p) {
            p.connected = true;
            peripherals.set(peripheral.id, p);
            this.setState({peripherals});
          }
          console.log('Connected to ' + peripheral.id);

          setTimeout(() => {
            // test: sending a command to the GATT bluFi server
            BleManager.retrieveServices(peripheral.id).then((peripheralInfo) => {
              console.log(peripheralInfo);
              var service = 'ffff'; 
              var toESP32Characteristic = '0000ff01-0000-1000-8000-00805f9b34fb' ; //'ff01';
              var wifiStatus = '14000100'; // '13333333-3333-3333-3333-333333330001';

              setTimeout(() => {
                BleManager.startNotification(peripheral.id, service, toESP32Characteristic).then(() => {
                  console.log('Started notification on ' + peripheral.id);
                  setTimeout(() => {
                    BleManager.write(peripheral.id, service, toESP32Characteristic, [0]).then(() => {
                      console.log('Writed toESP32Characteristic');
                      BleManager.write(peripheral.id, service, wifiStatus, [0]).then(() => {
                        console.log('Writed wifiStatus');
                        /*
                        var wifiStatus = {
 
                        };*/
                      });
                    });

                  }, 1500);
                }).catch((error) => {
                  console.log('Notification error', error);
                });
              }, 600);
            });

          }, 1900);
        }).catch((error) => {
          console.log('Connection error', error);
        });
```
### Stack trace and console log

JS Remote debugger console:
```
Connected to 24:0A:C4:1C:91:1E

App.js:200 {characteristics: Array(6), services: Array(3), advertising: {…}, rssi: -61, id: "24:0A:C4:1C:91:1E", …}advertising: {txPowerLevel: 3, serviceUUIDs: Array(1), isConnectable: true, manufacturerData: {…}, localName: "MLAB100"}characteristics: Array(6)0: {descriptors: Array(1), properties: {…}, characteristic: "2a05", service: "1801"}1: {properties: {…}, characteristic: "2a00", service: "1800"}2: {properties: {…}, characteristic: "2a01", service: "1800"}3: {properties: {…}, characteristic: "2aa6", service: "1800"}4: {properties: {…}, characteristic: "ff01", service: "ffff"}5: {descriptors: Array(1), properties: {…}, characteristic: "ff02", service: "ffff"}length: 6__proto__: Array(0)id: "24:0A:C4:1C:91:1E"name: "MLAB100"rssi: -61services: (3) [{…}, {…}, {…}]__proto__: Object

App.js:226 Notification error Set notification failed for 0000ff01-0000-1000-8000-00805f9b34fb
```

adb log:

```
02-21 10:12:50.925 20686  6790 D ReactNativeBleManager: Retrieve services from: 24:0A:C4:1C:91:1E
02-21 10:12:50.925 20686  6790 D BluetoothGatt: discoverServices() - device: 24:0A:C4:1C:91:1E
02-21 10:12:50.926  4931  4969 D BtGatt.GattService: discoverServices() - address=24:0A:C4:1C:91:1E, connId=7
02-21 10:12:50.926  4931  5083 E bt_btif_gattc: btif_gattc_upstreams_evt: Unhandled event (7)!
02-21 10:12:50.926  4931  5083 E bt_btif_gattc: btif_gattc_upstreams_evt: Unhandled event (7)!
02-21 10:12:50.926  4931  5083 E bt_btif_gattc: btif_gattc_upstreams_evt: Unhandled event (7)!
02-21 10:12:50.926  4931  5083 I bt_btif : HAL bt_gatt_callbacks->client->search_complete_cb
02-21 10:12:50.926  4931  5083 D BtGatt.GattService: onSearchCompleted() - connId=7, status=0
02-21 10:12:50.926  4931  5083 D bt_bta_gattc: bta_gattc_get_gatt_db
02-21 10:12:50.926  4931  5083 I bt_btif : HAL bt_gatt_callbacks->client->get_gatt_db_cb
02-21 10:12:50.927  4931  5083 D BtGatt.GattService: onGetGattDb() - address=24:0A:C4:1C:91:1E
02-21 10:12:50.927  4931  5083 D BtGatt.GattService: got service with UUID=00001801-0000-1000-8000-00805f9b34fb
02-21 10:12:50.927  4931  5083 D BtGatt.GattService: got characteristic with UUID=00002a05-0000-1000-8000-00805f9b34fb
02-21 10:12:50.927  4931  5083 D BtGatt.GattService: got descriptor with UUID=00002902-0000-1000-8000-00805f9b34fb
02-21 10:12:50.928  4931  5083 D BtGatt.GattService: got service with UUID=00001800-0000-1000-8000-00805f9b34fb
02-21 10:12:50.928  4931  5083 D BtGatt.GattService: got characteristic with UUID=00002a00-0000-1000-8000-00805f9b34fb
02-21 10:12:50.928  4931  5083 D BtGatt.GattService: got characteristic with UUID=00002a01-0000-1000-8000-00805f9b34fb
02-21 10:12:50.928  4931  5083 D BtGatt.GattService: got characteristic with UUID=00002aa6-0000-1000-8000-00805f9b34fb
02-21 10:12:50.928  4931  5083 D BtGatt.GattService: got service with UUID=0000ffff-0000-1000-8000-00805f9b34fb
02-21 10:12:50.928  4931  5083 D BtGatt.GattService: got characteristic with UUID=0000ff01-0000-1000-8000-00805f9b34fb
02-21 10:12:50.928  4931  5083 D BtGatt.GattService: got characteristic with UUID=0000ff02-0000-1000-8000-00805f9b34fb
02-21 10:12:50.928  4931  5083 D BtGatt.GattService: got descriptor with UUID=00002902-0000-1000-8000-00805f9b34fb
02-21 10:12:50.929 20686 20698 D BluetoothGatt: onSearchComplete() = Device=24:0A:C4:1C:91:1E Status=0
02-21 10:12:50.950  3040  3431 D audio_hw_primary: start_output_stream: exit
02-21 10:12:50.979 20686  6790 D ReactNativeBleManager: startNotification
02-21 10:12:50.980 20686  6790 D ReactNativeBleManager: registerNotify
02-21 10:12:50.980 20686  6790 D ReactNativeBleManager: setNotify
02-21 10:12:50.980 20686  6790 D BluetoothGatt: setCharacteristicNotification() - uuid: 0000ff01-0000-1000-8000-00805f9b34fb enable: true
02-21 10:12:50.982  4931 29006 D BtGatt.GattService: registerForNotification() - address=24:0A:C4:1C:91:1E enable: true
02-21 10:12:50.983  4931  5083 I bt_btif : HAL bt_gatt_callbacks->client->register_for_notification_cb
02-21 10:12:50.984  4931  5083 D BtGatt.GattService: onRegisterForNotifications() - address=null, status=0, registered=1, handle=42
```
There is a suspicious `address=null` in the BtGatt.GattService: onRegisterForNotifications().

Can someone please suggest how to get to the bottom of it? I have tried to add some delays (console.log calls) after BleManager.retrieveServices() with no luck.

Thanks.

marcoeg, 2019-02-21:
My bad the characteristic is a "Read" and not writeable.
In other words, you can get this error if using the characteristic improperly.

closed 2019-02-21 by marcoeg

cheetahteam355, 2020-10-23:
And how can you write?