innoveit/react-native-ble-manager #481: Can't find any device on iOS

Ilario17, 2019-02-28:
### Version
- react-native-ble-manager v6.6.2
- react-native v0.58.4
- iOS 12.1.4

The app can not find any devices, I've tried with Bluetooth Smart Discover app and it works.

My code:

    constructor(props) {
		super(props);

		this.handleDiscoverPeripheral = this.handleDiscoverPeripheral.bind(this);
		//this.handleUpdateValueForCharacteristic = this.handleUpdateValueForCharacteristic.bind(this);
	}

    componentDidMount() {
		BleManager.start({ showAlert: false })
			.then(() => {
				console.log('module ooook')
			});
        this.handlerDiscover = bleManagerEmitter.addListener('BleManagerDiscoverPeripheral', this.handleDiscoverPeripheral);

        BleManager.scan([], 20, false)
	   	.then(() => {
	   		console.log('scan started...');
	   	})

    //NEVER CALLED
    handleDiscoverPeripheral(peripheral) {
		console.log('handleDiscoverPeripheral....... ****** ');
		console.log(peripheral);
	}

marcosinigaglia, 2019-02-28:
Hi, did you try the example code?

Ilario17, 2019-02-28:
Hi, no I haven't tested it, but seems almost the same

kemicofa, 2019-03-20:
I'm getting the same issue. No device is showing on the iOS application, but works completely fine on Android.

kemicofa, 2019-03-20:
Just to confirm, I downloaded and ran the example and it works. I'll see what was wrong with what I did.

kemicofa, 2019-03-20:
So after implementing the example solution within my project it finally worked. It would appear that `this.handlerDiscover = bleManagerEmitter.addListener('BleManagerDiscoverPeripheral', this.handleDiscoverPeripheral);` does not work when placed in a Mobx Store... I'll keep checking to see if the store is the cause of this issue or not.

wentianl20, 2019-03-25:
Hi, guys, I have the same issue. How about setting up a timeout manually? Does it solve the problem?

closed 2019-06-18 by marcosinigaglia