innoveit/react-native-ble-manager #485: First write isn't working

tal987, 2019-03-11:
I'm trying to write into my device, when I'm using different app like NRFconnect its working with no problem, but here only 'Write3' and 'Write4' is working,  'Write1' and 'Write2' not working and I still see the previous write. im not getting any error what should be the problem:

      BleManager.retrieveServices(peripheralId).then((peripheralData) => {
    //change name
    BleManager.write(peripheralId, '0000af40-62bd-468a-9b17-3918fdafb86d', '0000af41-62bd-468a-9b17-3918fdafb86d', [10]).then(() => {
      console.log('Writed1');

      BleManager.write(peripheralId, '0000af40-62bd-468a-9b17-3918fdafb86d', '0000af42-62bd-468a-9b17-3918fdafb86d',[53]).then(() => {
        console.log('Writed2');
      }) .catch((error) => {
        // Failure code
        console.log(error);
      });

    }) .catch((error) => {
      // Failure code
      console.log(error);
    });
     })
     BleManager.retrieveServices(peripheralId).then((peripheralData) => {

       // change length
        BleManager.write(peripheralId, '0000af40-62bd-468a-9b17-3918fdafb86d', '0000af41-62bd-468a-9b17-3918fdafb86d', [11]).then(() => {
          console.log('Writed3');
    
          BleManager.write(peripheralId, '0000af40-62bd-468a-9b17-3918fdafb86d', '0000af42-62bd-468a-9b17-3918fdafb86d',[1] ).then(() => {
            console.log('Writed4');
          })
    
        })
      })

marcosinigaglia, 2019-03-11:
Hi, you have to queue the writes, you can't write simultaneously

closed 2019-03-11 by marcosinigaglia

tal987, 2019-03-11:
@marcoinigaglia what do you mean queue?
I can’t do write inside the then of the first write? Should I separate the write to 4 functions?

marcosinigaglia, 2019-03-15:
You have to move the write 3 and 4 inside the write 2 then.