innoveit/react-native-ble-manager #489: Byte array for sending data

tal987, 2019-03-13:
Hey,
I have a question about the Write function and the data that I should pass.
For instance if I want to send the word 'hello'. I should send this array:
[68,65,6c,6c,6f] I mean the hex convert of the string or it may be this array:
[104,101,108,108,111] = the decimal convert of the string "hello"?

Thank you for the answer

wangrenhai, 2019-03-14:
Thank you for your reminding, thanks to your plug-in, or my bluetooth printer function simply can not do out. You are selfless devotees, you are the best.

tal987, 2019-03-14:
@marcosinigaglia Do you have any idea?

Kevin-zyk, 2019-03-14:
I sent an array like [122] in the write method, but the format was not correct

tal987, 2019-03-14:
@kevin-zyk hows your answer related to my question?

marcosinigaglia, 2019-03-15:
Hi @tal987 , depends on you device protocol. 
( You can also write in hex directly [0x68,0x65,0x6c,0x6c,0x6f] )

closed 2019-03-18 by marcosinigaglia