innoveit/react-native-ble-manager #492: iOS cant connect device after remember device

pmcheng99, 2019-03-26:
### Version

Tell us which versions you are using:

- react-native-ble-manager v6.6.2
- react-native v0.59.1
- iOS/Android v.12.2

### Expected behaviour

First time i connect the ble device, iOS prompt password input dialog. After input password, connection succeed, and iOS device remember the ble device so that next time connection should not need password and succeed.

### Actual behaviour

First time is fine, but after that, hint will not prompt out and connection will failed.
Android device is OK.

### Steps to reproduce

1. Prepare a ble device which need password and can be remember.
2. Connect the device and input password.
3. Connect succeed and iOS device remember the ble device(check in iOS setting page)
4. Disconnect it and try to connect again.

### Stack trace and console log

Hint: it would help a lot if you enable the debugger ("Pause on exceptions" in the "Source" panel of Chrome dev tools) and spot the place where the error is thrown

no error log.

```
```

pmcheng99, 2019-03-26:
> ### Version
> Tell us which versions you are using:
> 
> * react-native-ble-manager v6.6.2
> * react-native v0.59.1
> * iOS/Android v.12.2
> 
> ### Expected behaviour
> First time i connect the ble device, iOS prompt password input dialog. After input password(pin code), connection succeed, and iOS device remember the ble device so that next time connection should not need password and succeed.
> 
> ### Actual behaviour
> First time is fine, but after that, hint will not prompt out and connection will failed.
> Android device is OK.
> 
> ### Steps to reproduce
> 1. Prepare a ble device which need password and can be remember.
> 2. Connect the device and input password.
> 3. Connect succeed and iOS device remember the ble device(check in iOS setting page)
> 4. Disconnect it and try to connect again.
> 
> ### Stack trace and console log
> Hint: it would help a lot if you enable the debugger ("Pause on exceptions" in the "Source" panel of Chrome dev tools) and spot the place where the error is thrown
> 
> no error log.

marcosinigaglia, 2019-03-26:
Hi @pmcheng99 , sorry but I never tested a secured connection. From the documentation is all at OS level but I can't help you.

closed 2019-06-18 by marcosinigaglia