innoveit/react-native-ble-manager #493: Be able to listen to BOND_BONDED state event

fkhosrow, 2019-03-31:
### Version

Tell us which versions you are using:
- iOS/Android v.? Android
 
### Steps to reproduce
1. Run an Android app.
2. Turn off a paired device.
3. Turn this device back on.
4. The App connects to this paired device by calling createBond() followed by connect(). 
5. The App at this point registers and listens to a NOTIFY characteristics on the paired device. **The paired device fails to start the read.** 

6. Restart the app but leave the pair device on.
7. The App connects to this paired device by calling createBond() followed by connect().
8. The App at this point registers and listens to a NOTIFY characteristics on the paired device. The paired device however is able to start the read and the notify callback is called. Success.

I believe the reason the start read fails is that a BOND_BONDED event is sent after setNotify is complete. This messes the Bluetooth stack. 

### Expected Behavior
The callback that was set in createBond() should be called such that the App can listen to this event and resets the Bluetooth stack. 

### Actual behaviour
The callback that was set in createBond() in BleManager.java should be called such that the App can listen to this event and resets the start notification callback after actual bonding is successful.

See BleManager.java createBond and BroadcastReceiver calls

### Stack trace and console log

marcosinigaglia, 2019-04-01:
Hi, @fkhosrow  I don't have a device to test, if you think your commit is ok make a PR.

fkhosrow, 2019-04-07:
Hi @marcosinigaglia no problem. I will test it some more on couple of phones and submit a PR this week.

fkhosrow, 2019-04-27:
@marcosinigaglia I created a PR for this purpose. Thanks.

closed 2019-04-27 by fkhosrow