innoveit/react-native-ble-manager #494: how to get the response data from startNotification()

xwilluhn, 2019-04-01:
### Version

Tell us which versions you are using:

- react-native-ble-manager v6.6.2
- react-native v0.58.6
- iOS

I need to read the response data (Write/Notify)
here is part of my code:

      const notifyResponse = await BleManager.startNotification(
        id,
        serviceUUID,
        characteristicUUID_Read
      );

      console.log("notify response data", notifyResponse);

      const writeReturn = await BleManager.write(
        id,
        serviceUUID,
        characteristicUUID,
        getVersionWrite
      );

I could see the response from :

  handleUpdateValueForCharacteristic(data) {
    console.log(' Received data from ' + data.peripheral + ' characteristic ' + data.characteristic);
    console.log('data', bytesToString(data.value))
  }

And I need to ask BLE device multiple times for different information. How to read these value out? 

With read(), I have no problem to get the value, I don't know how to get it with startNotification()

Thanks!

marcosinigaglia, 2019-04-01:
Hi, is an event, check the documentation and the example project.

xwilluhn, 2019-04-01:
I did. 
like here: https://github.com/innoveit/react-native-ble-manager#blemanagerdidupdatevalueforcharacteristic
It does console.log the data. still not sure how to read the data and use it later.

closed 2019-04-02 by xwilluhn

yoonachen, 2021-12-14:
Hey, do you solve this problem now?
I have the same issue and want to know how to do it.
Thanks.