innoveit/react-native-ble-manager #495: Can we handle more than one BleManagerDidUpdateValueForCharacteristic listener for different characteristics?

erdouzun, 2019-04-02:
- react-native-ble-manager v6.6.1
- react-native v.58
- iOS/Android v.12-v11-v10

### Expected behaviour

Hi everyone,

In my application, ble device has 4 characteristics. 2 for writing, 2 for reading.

I just add the listener "BleManagerDidUpdateValueForCharacteristic" for a reading characteristic.

Then I start the notification, and everything is ok.

The device second characteristic is for using mobile application as a proxy to access cloud. So I have to differ these listeners, and I try to add a new listener "BleManagerDidUpdateValueForCharacteristic" for 2nd reading characteristic.

Is it possible? If this possible, how can I differ the listeners for different characteristics? 

Thx in advance.

marcosinigaglia, 2019-04-02:
Hi, you have the characteristic in the event parameter (https://github.com/innoveit/react-native-ble-manager#blemanagerdidupdatevalueforcharacteristic)

erdouzun, 2019-04-03:
I just skipped the detail that listeners can have parameters "value, peripheral, characteristic, service". 

So this solves my problem. Thanks for your reply @marcosinigaglia

closed 2019-04-03 by erdouzun