innoveit/react-native-ble-manager #498: UpdateValueForCharacteristic receives data very slowly on android

pooja-chothe, 2019-04-15:
### Version

Tell us which versions you are using:

- react-native-ble-manager v6.6.2
- react-native v0.55.4
- iOS/Android v.?

I am using react-native-ble-manager module for bluetooth communication(BLE). When I write some command using bleManager.Write() function then device returns some amount of data, but data receiving speed is very slow on android and better on iOS. So how i can improve the data receiving speed. Please let me know if anyone know the solution.

Thanks in advance.

drchorney, 2019-04-20:
bandwidth is proportional to the MTU packet size, and inversely proportional to the event time duration. Need low packet duration, and large number of bytes for the MTU.

This a good one to read for some background on ble throughput. https://www.silabs.com/community/wireless/bluetooth/knowledge-base.entry.html/2015/08/06/throughput_with_blue-Wybp.

I'm having issues as well, I'm trying to increase the MTU size with bleManager.requestMTU(id, 244). The promise returns and says it has updated it, but I know it hasn't worked, as I can see on my embedded device the MTU is stuck at 20 bytes. @pooja-chothe I suspect you are having similar issues, and your MTU is stuck at 20 bytes as well ? You should try increasing it, and see if you have any luck. Sounds like iOS automatically negotiates as high MTU as possible. That's why your iOS is working good, and android like crap. 

Anyone have any luck successfully increasing the MTU on android ? I 've tried a different app, same phone, to same device, I'm getting MTU of 244. So I think it's an issue with this library ?

Thanks

JacopoKenzo, 2019-06-13:
Any update on this?
Having the same issue.

JacopoKenzo, 2019-06-18:
Doesn't look like a closed issue to me :)

drchorney, 2019-06-18:
@JacopoKenzo , unfortunately no, I tried playing around with various configuration/settings to try to get the MTU to actually work. But no success. In the end I switched over to react-native-ble-plx. When I set the MTU size with react-native-ble-plx, it actually works. So I'm not using this library anymore...

JacopoKenzo, 2019-06-18:
@drchorney I was having problems receiving notifications with react-native-ble-plx that's why I switched to this library.. wonder if I should give it another try.

drchorney, 2019-06-19:
@JacopoKenzo, yeah I started with react-native-ble-plx as well, and then I also switched over to this library as I liked the listener logic and the more functional nature of the library. I hate passing around the `manager = new BleManager()`. But.... I hate transferring data at an MTU of 20 bytes much much more... With react-native-ble-plx I can max out the MTU size on android.

marcosinigaglia, 2019-06-19:
Hi guys, I can reopen the issue but I can't test on your device and I don't see problem in the code.

danielalza, 2019-07-17:
yo tambien tengo problemas aumentando el MTU. no cambia de 20

craiq, 2020-04-01:
Hey I'm new to react native but just had a look at this as I like to use this library too.
I just checked [the android java code](https://github.com/innoveit/react-native-ble-manager/blob/master/android/src/main/java/it/innove/Peripheral.java) and it says on line 761 we need min API 21 for mtu requests.
Can you check if increasing the minsdk solves the problem?

Rowansdabomb, 2021-08-20:
This is what ended up working for me.

```  
const mtu = await BleManager.requestMTU(deviceID, 128); # request the MTU size in bytes
const services = await BleManager.retrieveServices(deviceID);
await BleManager.writeWithoutResponse(deviceID, 'D100', 'D101', API_KEY, mtu); # pass the MTU size in bytes to maxByteSize optional param
 ```

The `BleManger.write()` documentation mentions that `maxByteSize` is the number of bytes before the message is split, but `BleManager.writeWithoutResponse()` does not.

Relevant docs: https://github.com/innoveit/react-native-ble-manager#writeperipheralid-serviceuuid-characteristicuuid-data-maxbytesize

closed 2023-12-04 by marcosinigaglia

saqib-ahmed, 2025-02-09:
> This is what ended up working for me.
> 
> ```
> const mtu = await BleManager.requestMTU(deviceID, 128); # request the MTU size in bytes
> const services = await BleManager.retrieveServices(deviceID);
> await BleManager.writeWithoutResponse(deviceID, serviceId, characteristicId, data, mtu); # pass the MTU size in bytes to maxByteSize optional param
> ```
> 
> The `BleManger.write()` documentation mentions that `maxByteSize` is the number of bytes before the message is split, but `BleManager.writeWithoutResponse()` does not.
> 
> Relevant docs: https://github.com/innoveit/react-native-ble-manager#writeperipheralid-serviceuuid-characteristicuuid-data-maxbytesize

I had the same issue and this worked for me as well. I got it working with 517 MTU.