innoveit/react-native-ble-manager #500: didDiscoverCharacteristicsForService: Attempted to dereference null pointer

elliottkember, 2019-04-24:
### Version

Tell us which versions you are using:

- react-native-ble-manager v6.6.1
- react-native v0.57.1
- iOS/Android v12

### Expected behaviour

The application should not crash when a device is disconnected

### Actual behaviour

It seems to be that if I'm retrieving a characteristic for the connected BLE device, but the device is disconnected at the same time, it crashes with the above error.

### Steps to reproduce

1. Have device connected
2. Simultaneously power off the device while retrieving a characteristic
3. App crashes

### Stack trace and console log

Happens at BLEManager.m line 593 in `retrieveServices:(NSString *)deviceUUID services:(NSArray<NSString *> *)services callback:(nonnull RCTResponseSenderBlock)callback`:

`[retrieveServicesCallbacks setObject:callback forKey:[peripheral uuidAsString]];`

```2019-04-23 20:45:13.768647-0700 Soulmate 🐛[1158:761779] Service 4FAFC201-1FB5-459E-8FCC-C5C9C331914B <CBService: 0x281223cc0, isPrimary = YES, UUID = 4FAFC201-1FB5-459E-8FCC-C5C9C331914B>
2019-04-23 20:45:13.769748-0700 Soulmate 🐛[1158:761779] Characteristics For Service Discover
2019-04-23 20:45:13.771732-0700 Soulmate 🐛[1158:770057] retrieveServices (null)
2019-04-23 20:57:29.454065-0700 Soulmate 🐛[1158:771242] XPC connection interrupted
```

`Thread 550: EXC_BAD_ACCESS (code=1, address=0x3332363133354424)`

marcosinigaglia, 2019-05-03:
Hi @elliottkember , do you have the crash only on iOS?

elliottkember, 2019-05-03:
@marcosinigaglia Hi! Yes I'm only building for iOS with this app.

closed 2019-06-18 by marcosinigaglia

iradkot, 2019-12-30:
Hey, why was this close?