innoveit/react-native-ble-manager #503: Android doesn't discover any devices after a small while

wesdewitte, 2019-05-02:
### Version

Tell us which versions you are using:

- react-native-ble-manager v6.6.2
- react-native v0.59.5
- Android 8.1.0

### Expected behaviour

Im scanning for 10 seconds long every 50 seconds. it should scan devices.

### Actual behaviour

it scans devices till about the 5th time around that time. Then it doesn't receive any devices anymore. Situation doesn't change is the meanwhile. The handleDiscoverPeripheral listener doesn't get any devices anymore.

### Steps to reproduce
// Bluetooth Manager
import BleManager from 'react-native-ble-manager';
BleManagerModule = NativeModules.BleManager;
bleManagerEmitter = new NativeEventEmitter(BleManagerModule);

1. BleManager.start();
2. scan 10 seconds long for every 50 seconds BleManager.scan([], 10, false)
3.bleManagerEmitter.addListener('BleManagerDiscoverPeripheral', this.handleDiscoverPeripheral );

wesdewitte, 2019-05-02:
In the logs i came across this : 

`2019-05-02 10:54:48.074  D/BluetoothLeScanner: Start Scan with callback
2019-05-02 10:54:48.077  D/BluetoothLeScanner: onScannerRegistered() - status=133 scannerId=0 mScannerId=0
2019-05-02 10:54:48.077  D/BluetoothLeScanner: Registration failed, unregister scannerId = 0
2019-05-02 10:54:48.077 D/BluetoothLeScanner: Scan failed, reason: app registration failed`

marcosinigaglia, 2019-06-18:
Hi, do you find something?

wesdewitte, 2019-06-25:
I think because android is blocking the scanning

``` 2019-06-25 12:02:14.499 4333-4807/? W/BtGatt.ScanManager: Cannot start unfiltered scan in screen-off. This scan will be resumed later: 8
2019-06-25 12:03:14.766 4333-4807/? W/BtGatt.ScanManager: Cannot start unfiltered scan in screen-off. This scan will be resumed later: 8
2019-06-25 12:04:05.633 4333-4807/? W/BtGatt.ScanManager: Cannot start unfiltered scan in screen-off. This scan will be resumed later: 8
2019-06-25 12:05:37.789 4333-4807/? W/BtGatt.ScanManager: Cannot start unfiltered scan in screen-off. This scan will be resumed later: 8
2019-06-25 12:07:00.289 4333-4807/? W/BtGatt.ScanManager: Cannot start unfiltered scan in screen-off. This scan will be resumed later: 8 ```

Hassam34, 2019-12-23:
### Just extend the time of searching. In my case i extend time from 3 second to 5 and it is searching approximately all the available 
 BleManager.scan([], 5, true).then((results) => {
                console.log('Scanning...');
                this.alreadySent = []
                this.setState({ scanning: true, show: false });
            });

quangcanh2975, 2020-03-24:
As same as in my case after upgrading react native from version 0.57 to 0.61. At first, app can scan devices very well, but after about 30 seconds, the scan event return no devices.

quangcanh2975, 2020-03-24:
> I think android is blocking the scanning so it returns no devices
> 
> ```
> 2019-06-25 12:03:14.766 4333-4807/? W/BtGatt.ScanManager: Cannot start unfiltered scan in screen-off. This scan will be resumed later: 8
> 2019-06-25 12:04:05.633 4333-4807/? W/BtGatt.ScanManager: Cannot start unfiltered scan in screen-off. This scan will be resumed later: 8
> 2019-06-25 12:05:37.789 4333-4807/? W/BtGatt.ScanManager: Cannot start unfiltered scan in screen-off. This scan will be resumed later: 8
> 2019-06-25 12:07:00.289 4333-4807/? W/BtGatt.ScanManager: Cannot start unfiltered scan in screen-off. This scan will be resumed later: 8 ```
> ```
Did you find out something?

fbambusi, 2020-05-17:
Hi, I am experiencing the same problem on Android 10. I could not solve the problem, so I am using a workaround.

In my use case the user is supposed to know the peripheral to which the application should connect: as a consequence, he/she should leave the application and manually pair the peripheral.

At this point, the peripheral will be available using
`BleManager.getBondedPeripherals([])`

Looking forward to find a final solution.

joy-mollick, 2021-09-16:
This package is not working one. This is not right choice to use this package for bluetooth tasks with react native . Don't run for this , there are more other pacakges all of you can use .

marcosinigaglia, 2021-09-17:
Hi @joy-mollick , I don't think your comment is useful, the library is working in many situations, if you find a bug you can solve it and make a PR.

closed 2023-02-10 by marcosinigaglia