innoveit/react-native-ble-manager #504: Notifications not working on Android

MarkBlythe, 2019-05-03:
### Version

Tell us which versions you are using:

- react-native-ble-manager v6.6.4
- react-native v0.59.6
- iOS 12 / Android 7.1.2
 
### Expected behaviour

1. Connect to blood pressure machine.
2. Retrieve Services
3. Start notification for blood pressure reading
4. Receive data from machine in BleManagerDidUpdateValueForCharacteristic 

### Actual behaviour

iOS works perfectly as described above.
Android devices can connect and retrieve services but `BleManager.startNotification` doesn't seem to be doing anything.

### Steps to reproduce

1. Update handler set in componentDidMount
`this.handlerUpdate = bleManagerEmitter.addListener('BleManagerDidUpdateValueForCharacteristic', this.handleUpdateValueForCharacteristic);`

2. Perform a scan of bluetooth devices (This works properly)
3. Connect and start notification

```
BleManager.connect(peripheral).then(() => {

            setTimeout(() => {

                BleManager.retrieveServices(peripheral).then((peripheralInfo) => {
                    console.log(peripheralInfo);

                    setTimeout(() => {
                        BleManager.startNotification(peripheralId, serviceUuid, characteristicUuid).then(() => {
                            console.log('Notification started for Service: ' + serviceUuid + ' and characteristic: ' + characteristicUuid);
                        }).catch((error) => {
                            console.log(error);
                        });
                    }, 1000);
                });

            }, 1000);

        }).catch((error) => {
            console.log(error);
        });
```
4. Wait for `BleManagerDidUpdateValueForCharacteristic` which is never called.

### Stack trace and console log

This is the characteristic in question: 
`
characteristic: "2a35" 
descriptors: [{…}]
properties: {Indicate: "Indicate"}
service: "1810"
`

`Notification started for Service: 1810 and characteristic: 2a35`
`Disconnected from B0:49:5F:02:AA:4D`

marcosinigaglia, 2019-05-03:
Hi, you have to debug directly on your device, also try other smarphone with different android version.

MarkBlythe, 2019-05-08:
After some more testing I realised it is because in Android you need to pair the device externally before trying to gather data from it. 

iOS handles the pairing automatically in the connect method but there is no similar behaviour for Android devices. 

In summary, to be able to connect and read / receive data from Android devices the peripheral must already be paired with the device outside of your application.

closed 2019-05-08 by MarkBlythe