innoveit/react-native-ble-manager #505: Is receiving data in a separate thread from JS thread?

KongfuMan, 2019-05-05:
### Version

Tell us which versions you are using:

- react-native-ble-manager v6.6.2
- react-native v0.57.8 
- iOS/Android v26.0.1

### Expected behaviour

When I tried to receive data from a sensor connected with android mobile device, do some real-time calculation, and present the result on the phone screen, I found a terrible delay. My algorithm is pretty easy, it shouldn't be a problem. So I am wondering if the receiving data is done in the separate thread from the js thread ? 

### Actual behaviour

Tell us what happens instead

### Steps to reproduce

1.
2.
3.

### Stack trace and console log

Hint: it would help a lot if you enable the debugger ("Pause on exceptions" in the "Source" panel of Chrome dev tools) and spot the place where the error is thrown

```
```

marcosinigaglia, 2019-05-06:
Hi, the ble protocol is not "fast", how long is your delay?

KongfuMan, 2019-05-06:
> Hi, the ble protocol is not "fast", how long is your delay?

About 200~300ms delay. I am wondering if the delay caused by the receiving data or updating the redux store?

marcosinigaglia, 2019-05-06:
I think is possible that the react native bridge and redux layer can together add that delay.
Check the log in the device to have a better idea.

closed 2019-06-18 by marcosinigaglia