innoveit/react-native-ble-manager #506: getConnectedPeripherals type error

OrLevy23, 2019-05-05:
### Version

Tell us which versions you are using:

- react-native-ble-manager v6.6.1
- react-native v0.58
- iOS/Android v. iOS

### Expected behaviour
getConnectedPeripherals has wrong return type (typescript)
the return value suppose promise with array of peripherals but instead 

```
```

wangrenhai, 2019-05-05:
Dear author,

I would like to ask why once my SDK version exceeds 22, bluetooth can not be searched, I have been very troubled, if you can help me solve, I am very grateful...

marcosinigaglia, 2019-05-06:
Hi @OrLevy23 , can you check if you have the error only when there aren't connected peripherals?

OrLevy23, 2019-05-06:
@marcosinigaglia, the problem is with the types (TS) of the native module, when looked at the native module i saw that the return value is callback of array of peripherals, but the types file says that the return value is ``` Promise<void>```

marcosinigaglia, 2019-05-06:
I don't use TS but in https://github.com/innoveit/react-native-ble-manager/blob/master/index.d.ts file I see `Promise<any[]>`.

closed 2019-06-18 by marcosinigaglia