innoveit/react-native-ble-manager #508: stop scan delay

jazzymb7, 2019-05-08:
### Version

Tell us which versions you are using:

- react-native-ble-manager v6.6.4
- react-native v0.59.2
- iOS/Android v28

### Expected behaviour

Stop scan function should immediately be called after the designated number of seconds given in the scan function.

### Actual behaviour

Call scan function for 3 seconds, if you have more than 40-50 devices around you then the stop function will not be called until 10 seconds.

### Steps to reproduce

1. Use empty array as first argument in the scan function
2. Use 3 seconds as the second argument
3. Stop scan will not be called after 3 seconds

### Stack trace and console log

Hint: Use time stamps as to when the scan is being called and stop is being called, you'll see significant delay in stop scan.

marcosinigaglia, 2019-05-13:
Hi, I can't reproduce it, I think the delay is caused by the time needed to check the peripherals services for the filter. If you find a solution please make a PR.

jazzymb7, 2019-05-13:
Yes. If there is a filter on which the search has to be made that is the cause. But if no search criteria is provided then If more than 50 peripherals are found, I’m getting 50 callbacks each for one result. How do I control the callback?

marcosinigaglia, 2019-05-13:
Do you have 50 callbacks? In which function?

jazzymb7, 2019-05-13:
The one which gives results.

Connor-Y-Evenset, 2019-05-13:
You might want to benchmark performance on your device while running the scan function. JS has some troubles with keeping track of time when under significant load. If you don't want to benchmark, an easy check is to use `setTimeout` and see if it resolves in the correct amount of time.

jazzymb7, 2019-05-13:
I used timeout. But it is called only after all the results for the scan period are returned. Shouldn’t JavaScript call the timeout function immediately?

Connor-Y-Evenset, 2019-05-13:
Just to confirm my understanding is correct, you have something like
```
// Stop Call Handler
function handleStopScan() {
  console.log(new Date().toISOString());
}

// Add stop listener
bleManagerEmitter.addListener('BleManagerStopScan', this.handleStopScan);

// Start Scan
console.log(new Date().toISOString());
await BleManager.scan(['some id'], 3, false);
```
And the time printed by `handleStopScan()` is ~10s after when the scan started correct?

jazzymb7, 2019-05-13:
Yes. That is correct. Remove ‘some id’ and scan. That way you can test my scenario of more than 20-30 devices.

Connor-Y-Evenset, 2019-05-13:
I don't currently have 30 of my own devices to test with right now, but I ran the above snippet (without the id) and it was able to pickup 25 nearby devices (no idea whose they are) within an average around ~3.2 seconds. Are you running any logic in the `BleManagerDiscoverPeripheral` event listener?

jazzymb7, 2019-05-13:
Nope. No logic. You have the issue reproduced. You scanned for 3 but got back the results in 3.2. As the number of devices increases the delay in that time increases. Settimeout is not instantly calling. Any ideas?

Connor-Y-Evenset, 2019-05-13:
Interesting. There is a massive gap between an extra 0.2 seconds and the 8 seconds you reported though. Would it be possible for you to disable some of your transmitting devices until you drop down to 25 and see how long that takes you? I assumed that delay I had was due to the fact I'm running my app in dev mode and printing a bunch of stuff to the debug console. Let me try building for production and see if that makes a difference.

Connor-Y-Evenset, 2019-05-13:
Running the app in production without logs didn't seem to make a difference.

I was able to and scrounge up some more bluetooth enabled devices to be able to get up to 40 and the time taken didn't seem to increase still sitting at around 3.2s so it doesn't look like I've managed to get up to the 8s delay you reported.

For reference I'm running the tests using a Samsung S7 on API level 26

jazzymb7, 2019-05-14:
Here is the scenario
`
function scan(){ 
console.log(new Date().toISOString()) for example: 12:03 (only posting time)
BleManager.scan([],3,true)
}

//handle peripherals listener
bleManagerEmitter.addListener('BleManagerDiscoverPeripheral', this.handleDiscoverPeripheral);
//handle stop listener
bleManagerEmitter.addListener('BleManagerStopScan', this.stopScan);

//stop scan function
stopScan(){
console.log(new Date().toISOString()) 
 *****time here should be 12:06, but for me it goes to 12:11 or more in some cases******
}

//handle peripherals function
function handleDiscoverPeripherals(peripheral){
let x = this.state.data;
x.push(peripheral)
this.setState({data: x})
}
`

Connor-Y-Evenset, 2019-05-14:
Sorry, something has come up and I don't have time to continue testing this now. I'd recommend trying on a few different devices to see if that has any impact on the delay. If not hopefully someone else can help you or I'll try to get back to you when I have time (probably next week).

If you can leave your findings here I'll take a look at them as well

closed 2023-02-10 by marcosinigaglia

AmitNeuhaus, 2023-09-29:
Hey I am encountering the same issue,
Was there any solution or understanding for the source?

For the same code in android kotlin + name filter in not encountering any issues.