innoveit/react-native-ble-manager #510: Scan Option

keyz23, 2019-05-13:
I guys i'm using this library for my Android Application.
I have a question. 
I need to connect to a particular device (actually two with a very similar name, one right and one left).
I would like to know if there is a way to connect directly to a device without considering all the others.
I saw that in the function:
```
BleManager.scan ([], 5, true)
             .then (results => {
```

the results variable returns all the devices, how do I get back only what interests me?
Thanks a lot.

marcosinigaglia, 2019-05-14:
You can filter only by the service UUID, or you can connect directly to the peripheral id if you already saved it.

keyz23, 2019-05-15:
So can I connect automatically only if I had connected to this device once before?

marcosinigaglia, 2019-05-15:
Yes, to have the peripheral id you need to scan it one time.

keyz23, 2019-05-15:
To be sure that I understand what you're telling me, if I know the name of the device, can't I make a connection automatically if I haven't made a manual connection before? (To be clear if first I don't physically click on the Connect button)

marcosinigaglia, 2019-05-15:
To be precise: to connect you need the peripheral id, to have the peripheral id you need to scan the device at least one time.

closed 2019-06-18 by marcosinigaglia