innoveit/react-native-ble-manager #515: Keep recieved data in native code

a1lu, 2019-05-21:
### Version

Tell us which versions you are using:

- react-native-ble-manager v6.6.4
- react-native v0.59.8
- iOS/Android v.7

Hi,
thanks for your awesome work here. My question is not 100% related to your lib, but maybe you have an idea or a hint how to patch your lib for my case.
I use the lib to get data from my server peripheral using notifications. I put this data in my own native module.
 The datarate is about 6kbyte/s, on fast devices this works, though on older ones the react-native bridge is used to capacity. Do you know any way to communicate between two native modules? I could use some IPC mechanism, but I am not sure if thats the best way.

Best regards

marcosinigaglia, 2019-05-22:
Hi, I never did something similar, probably the easier way is to create a memory buffer to store the bytes you read.

a1lu, 2019-05-22:
My current implementation uses a BlockingQueue as Singleton. That works fine for the moment and reduces the cpu load by some percent.  Though I send only the notification events through this pipe. I don't need the other events like disconnected etc. in native, but it would be nice to have a good implementation upstream that one can use to subscribe all events in native code, like the useNativeDriver flag for react native animations. 

I am not an android / ios expert, so I am not sure whats the best way.
For Android there is a Looper / Handler thing:
https://developer.android.com/training/multiple-threads/communicate-ui#java

closed 2019-06-18 by marcosinigaglia