innoveit/react-native-ble-manager #521: Connection and getConnected functions error

OzgunAkkoyun, 2019-06-21:
### Version

Tell us which versions you are using:

- react-native-ble-manager: "^6.6.6"
- react-native : "0.59.9"

### Expected behaviour

I trying to only specific device which is bluetooth headphone,but i cant get any connected device or connect this device 

### Actual behaviour

its giving me error when i get connected devices "No connected peripherals" or connect function with mac adress. Other bounded functions all work fine.

marcosinigaglia, 2019-06-21:
Hi, I don't think you can get the connection of a bounded device on iOS, on Android use that functions..

OzgunAkkoyun, 2019-06-21:
Im using in android but i cant get any response in this functions.

rakshitbharat, 2023-11-22:
global.isBLEManagerInitialized = false;
if (!global.isBLEManagerInitialized) {
  BleManager.start({showAlert: false}).then(() => {
    console.error('Module initialized');
    global.isBLEManagerInitialized = true;
  });
}

add this in index.js and make sure BleManager.start will never be called again anywhere in app

closed 2023-12-04 by marcosinigaglia