innoveit/react-native-ble-manager #522: Duplicate received data when i turn off bt on my phone and reconnect

thuantvc, 2019-06-27:
In first connected, i'm receive data when i call BleManagerDidUpdateValueForCharacteristic is OK.
But when i turn off BT on my phone, after that i turn on, reconnect and receive duplicate value on notify

marcosinigaglia, 2019-06-28:
Probably you add the listener multiple time.

closed 2019-06-28 by marcosinigaglia

abdul-rightapps, 2020-04-22:
Same problem, i'm also facing, Any expert advice would be helpful.

Thanks in advance.

I do remove the handler/listener before creating it, also i have set a timer to do it,

Still when i turn off bluetooth on my phone and reconnect, I have 2 listeners.

My Code:

setTimeout(async () => {
            try {
              this.removeBleHandler();
            } catch (e) {
              console.log('error', e);
            }

            this.startBleHandler();
            stores.ble.startScan();
          }, 2000);

  startBleHandler() {
    this.handlerDiscover = bleManagerEmitter.addListener(
      'BleManagerDiscoverPeripheral',
      this.handleDiscoverPeripheral,
    );
    this.handlerStop = bleManagerEmitter.addListener(
      'BleManagerStopScan',
      this.handleStopScan,
    );

    // Run the Listener only once
    this.handlerUpdate = bleManagerEmitter.addListener(
      'BleManagerDidUpdateValueForCharacteristic',
      this.handleUpdateValueForCharacteristic,
    );

    console.log(
      'Run the Listener only once BleManagerDidUpdateValueForCharacteristic',
    );

    this.handlerDisconnect = bleManagerEmitter.addListener(
      'BleManagerDisconnectPeripheral',
      this.handleDisconnectedPeripheral,
    );
  }

  removeBleHandler() {
    this.handlerDiscover.remove();
    this.handlerStop.remove();
    this.handlerDisconnect.remove();
    this.handlerUpdate.remove(); // Don't Remove the listener as it required to listen for the button in background
  }