innoveit/react-native-ble-manager #523: You're already writing error on Android

zupaox, 2019-06-28:
### Version

Tell us which versions you are using:

- react-native-ble-manager ~v6.5.6
- react-native v0.59.5
- Android 9

### Expected behaviour

While concurrently fire `bleManager.write()` method, get `You're already writing` error on Android, but iOS works fine.
Benchmarked with iOS, concurrent write 20 messages takes 3-5 milliseconds, but sequentially it takes more than 2 seconds, which indicates a huge performance difference.
Expected Android to be able to concurrently write messages as well, if it is not due to limitation on OS or hardware.

### Actual behaviour

The library prevent the concurrently write on Android platform.

### Steps to reproduce

1. concurrently call `bleManager.write()` method.

### Questions
Is the constraint introduced by the library code or is it because the Android OS not supporting it?

marcosinigaglia, 2019-06-28:
Are you writing on different devices?

zupaox, 2019-06-29:
> Are you writing on different devices?

hi @marcosinigaglia: no the testing is done on same device, i.e. same peripheral.

marcosinigaglia, 2019-07-01:
You can't write concurrently, iOS is faster so probably you don't raise the error.

closed 2019-07-01 by marcosinigaglia