innoveit/react-native-ble-manager #525: Cancel connect to BLE device

shayhz, 2019-06-30:
### Version

Tell us which versions you are using:

- react-native-ble-manager v6.6.4
- react-native v0.50.3
- iOS/Android v. Android v9

### Expected behaviour

I want to implement connect with retries to ble device.
In order to do it, I start timeout for X seconds and then call to connect method.
In case the ble device is not in range, on some mobile apps, it takes time for connect to return an error. So, timeout method is called and reject the first attempt of connecting to the device.
Then, I want to start the second attempt to connect, but I can't call to "connect" method again, because it's already running since the seconds attempt.
I tried to call disconnect - maybe it will stop the connect method, but it didn't work.
I expect to have a way to cancel connect attempt.

### Actual behaviour
connect may take minutes to return error.

marcosinigaglia, 2019-07-01:
Hi, I think in iOS you can use [this function](https://developer.apple.com/documentation/corebluetooth/cbcentralmanager/1518952-cancelperipheralconnection).
In Android I thinks is not possible.

shayhz, 2019-07-01:
Thanks. but it means that I will need to fork the project and I don't want to :(
But I will need in case there is no other solution...

marcosinigaglia, 2019-07-01:
You can make a PR creating a new method in the library.

xJkit, 2019-11-05:
This feature is essential! Please help 😢

franzejr, 2021-04-17:
@shayhz do you think it makes sense to scan with retries rather than connect with retries? This is what I was trying to do. 

@marcosinigaglia one quick question for you: once we get the characteristics and the device information,  to connect to it, do we need to scan every time? As far as I can see on my tests, we always need to scan... which I thought it was not ideal. I'm wondering if with the device information I could just connect it directly without any scan.

closed 2023-12-11 by marcosinigaglia