innoveit/react-native-ble-manager #527: retrieveServices error "Device not connected"

mtumiati, 2019-07-12:
### Version

Tell us which versions you are using:

- react-native-ble-manager v6.6.6
- react-native v0.56
- Android 8.0

### Expected behaviour
after connecting to a device BleManager.retrieveServices(device.id) should not fail with error "Device not connected"

### Actual behaviour

``javascript
   async function connectToDevice(device){
      try {
            await BleManager.connect(device.id);

            const retServices = await BleManager.retrieveServices(device.id);

            //some other stuff
      }catch(err){
           console.log("error", err);
      }
   }

### Steps to reproduce

1. Connect to peripheral
2. call BleManager.retrieveServices on the connected peripheral

marcosinigaglia, 2019-07-12:
Hi, try to add a delay after the connection.

mtumiati, 2019-07-12:
I tried to add 2 seconds delay without success

marcosinigaglia, 2019-07-12:
So you have to debug directly in the native code, sorry but I can't reproduce your issue.

oscar-gallog, 2020-02-26:
I'm having the same issue, retrieveServices function never runs, even if you put delays.