innoveit/react-native-ble-manager #529: [Android] Peripheral not found when retrieving Services

BaloMueller, 2019-07-12:
### Version

Tell us which versions you are using:

- react-native-ble-manager v6.6.8
- react-native v59.8
- iOS/Android Android v.25

### Expected behaviour

Should return peripheralInfo, as it always did, and still does on iOS.

### Actual behaviour

Returns "Peripheral not found"

### Steps to reproduce

1. Connect to device
2. Call retrieveServices

### Stack trace and console log

I debugged already the Android code, but can't find the reason, why it is not working. The peripherals Map<> is always empty. When a peripheral gets discovered, it is added to the peripherals Map. But when retrieveServices() gets invoked, the Map is always empty. Can someone reproduce the issue?

```
```

marcosinigaglia, 2019-07-13:
Maybe is something related to this [commit](https://github.com/innoveit/react-native-ble-manager/commit/433db5d44ada2069406b9484294a552ba7f48612) . Do you use the `forceLegacy ` flag?

BaloMueller, 2019-07-13:
I agree, it is probably related to the commit. Are you able to reproduce?
No I don't use the forceLegacy flag.

marcosinigaglia, 2019-07-13:
No, I tested on different Android with no problem.

BaloMueller, 2019-07-14:
forceLegacy doesn't fix the problem.

BaloMueller, 2019-07-16:
Unfortunately I'm not experienced enough with the Android Platform to track the root cause down. I downgraded to Version 6.6.5, which solved the problem for me.

mtumiati, 2019-08-21:
I'm having the same issue on a Galaxy Note 9 running Android 9. My previous app version was using React Native 0.56 and react-native-ble-manager 6.6.2 and it was working. Since I updated to React Native 0.60 it's not working anymore even with 6.6.2 or 6.6.8. I'm going to do some checks hope to help.

mtumiati, 2019-08-22:
@BaloMueller I found the root cause for my case.  
I was calling startScan at every scan stop event if no devices was already connected. 
The problem was that the setState I was calling right after the connection was not fast enough to prevent the startScan to be called again. 
As I can see from the native code the "peripherals" map will be empty whenever a new scan starts

marcosinigaglia, 2019-08-23:
I think is better to use the BleManagerDiscoverPeripheral event to save the peripherals in your app.

mtumiati, 2019-08-23:
> I think is better to use the BleManagerDiscoverPeripheral event to save the peripherals in your app.

I'm using it. But I'm scanning and searching for a specific device. If I don't find it I need to rescan.

NeilMcGuiness, 2020-06-29:
> @BaloMueller I found the root cause for my case.
> I was calling startScan at every scan stop event if no devices was already connected.
> The problem was that the setState I was calling right after the [...]

Thank you so much for this, this just really helped us out. This should be in the documentation of the scan or retrieveServices function, as many people are likely performing continuously looping scans which will cause problems here.