innoveit/react-native-ble-manager #531: Continuous write failure?

1442916418, 2019-07-16:
Why Write Failure?

```
   write(data, index = 0) {
        data = stringToBytes(data);

        return new Promise((resolve, reject) => {
            BleManager.write(this.peripheralId, this.nofityServiceUUID[index], this.nofityCharacteristicUUID[index], data)
                .then((result) => {
                    console.log('success: ', result);
                    resolve();
                })
                .catch((error) => {
                    console.log('fail: ', error);
                    reject(error);
                });
        });
    
```

Console Printing:
Write failed

What's wrong? Please answer, thank you!

BaloMueller, 2019-07-16:
Please use the template to submit issues. What Version are you using? iOS or Android? 

Shouldn't the "Console Printing" be: "fail: Write failed"? Or is it just "Write failed"?

Have you executed retrieveServices() before for this peripheral? https://github.com/innoveit/react-native-ble-manager#retrieveservicesperipheralid-serviceuuids

1442916418, 2019-07-17:
Android,retrieveServices() executed.

Now there are new problems.

How to distinguish whether the currently connected Bluetooth UUID is readable or writable? Do you have any documents to refer to?

closed 2019-12-04 by marcosinigaglia