innoveit/react-native-ble-manager #533: Can not scan a speaker Bluetooth

nlavuit, 2019-07-25:
### Version

Tell us which versions you are using:

- react-native-ble-manager v^6.6.0
- react-native v^0.59.9
- iOS/Android

### Expected behaviour

I want to scan and connect with speaker Bluetooth from app

### Actual behaviour

I run an example and touch scan button, it scanned the devices such as smartTV, laptop, phone,... and them appear in a listview, but can not scan the speakers Bluetooth. I can scan these speakers in Bluetooth setting of my phone and connect success, but in the example app is not. Please help me resolve this issue, thanks.

marcosinigaglia, 2019-07-25:
Probably is not a BLE device.

nlavuit, 2019-07-25:
Hi @marcosinigaglia, I see info of speaker is Bluetooth 4.0, is it a BLE device?
I give you a link of the product:
https://www.anker.my/products/a3104-soundcore-nano-super-portable-wireless-bluetooth-speaker

marcosinigaglia, 2019-07-28:
Is not indicated that is a Bluetooth LE device.

closed 2019-07-28 by marcosinigaglia