innoveit/react-native-ble-manager #538: Do i get the major and minor from the "advertising.manufacturerData.bytes(bit array)" or "advertising.manufacturerData.data(base 64)"?

yang200258, 2019-08-02:
### Version

Tell us which versions you are using:

- react-native-ble-manager v?.?.?
- react-native v?.?
- iOS/Android v.?

### Expected behaviour
**_I want to get the major and minor from advertising data,but always has strange code,I want to know can i get it from advertising data??_** Any help will be grateful！！
[image]
Tell us what should happen

### Actual behaviour

Tell us what happens instead

### Steps to reproduce

1.
2.
3.

### Stack trace and console log

Hint: it would help a lot if you enable the debugger ("Pause on exceptions" in the "Source" panel of Chrome dev tools) and spot the place where the error is thrown

marcosinigaglia, 2019-08-05:
I don't understand your question.

yang200258, 2019-08-06:
> I don't understand your question.

Our company has a Bluetooth device that needs to obtain the data of major, minor, and UUID of each Bluetooth device when the phone scans to the Bluetooth list by use “BleManagerDiscoverPeripheral”. The device data scanned by my reactive-native is similar to the format of the screenshot. , there is no major and minor, I would like to ask can i  get data such as major and minor in the advertising data

yang200258, 2019-08-08:
I got major,minor by bytes to HexString,so i close this issue.

closed 2019-08-08 by yang200258

singh100ful, 2020-03-27:
> I got major,minor by bytes to HexString,so i close this issue.

[image]

I tried converting the bytes to hexString but got weird string above (see the first line)
I set major to 5 and minor to 65 using the ble simulator
Can you help me out here?

puranjayjain, 2023-03-01:
Same for me

puranjayjain, 2023-03-01:
@yang200258 can you give a short example if possible?

yang200258, 2023-03-01:
> @yang200258 can you give a short example if possible?

I am so sorry，it is been long time, I do not have my code, so i forgot how to deal this problem.