innoveit/react-native-ble-manager #540: iOS scanning not working..on iOS scanning can be started and stopped but it is not giving any scan result

shilpa108, 2019-08-07:
### Version

Tell us which versions you are using:

- react-native-ble-manager v?.?.?
- react-native v?.?
- iOS/Android v.?

### Expected behaviour

Tell us what should happen

### Actual behaviour

Tell us what happens instead

### Steps to reproduce

1.
2.
3.

### Stack trace and console log

Hint: it would help a lot if you enable the debugger ("Pause on exceptions" in the "Source" panel of Chrome dev tools) and spot the place where the error is thrown

```
```

shilpa108, 2019-08-09:
I am able to Start Ble manger , start and stop scanning. but it is not giving any scanning result in iOS. On android it is working fine.Please help

marcosinigaglia, 2019-08-14:
Check the installation and the app permission.

closed 2019-08-29 by marcosinigaglia