innoveit/react-native-ble-manager #542: App crashes when advertisement data seems to be null

folofse, 2019-08-22:
### Version

- react-native-ble-manager v6.6.8
- react-native v0.60.4
- iOS/Android: iOS 12.4

I got this error from Bugsnag from the production version of my app. It seems like RSSI or the Advertisement data is null and the app crashes.

```
NSInvalidArgumentException *** -[__NSPlaceholderDictionary initWithObjects:forKeys:count:]: attempt to insert nil object from objects[2] 
    Frameworks/CoreFoundation.framework/CoreFoundation ___exceptionPreprocess
    /usr/lib/libobjc.A.dylib _objc_exception_throw
    Frameworks/CoreFoundation.framework/CoreFoundation __CFThrowFormattedException
    Frameworks/CoreFoundation.framework/CoreFoundation -[__NSPlaceholderDictionary initWithObjects:forKeys:count:]
    Frameworks/CoreFoundation.framework/CoreFoundation +[NSDictionary dictionaryWithObjects:forKeys:count:]
    node_modules/react-native-ble-manager/ios/CBPeripheral+Extensions.m:280:10 dataToArrayBuffer
    node_modules/react-native-ble-manager/ios/CBPeripheral+Extensions.m:114:30 -[CBPeripheral(com_megster_ble_extension) serializableAdvertisementData:]
    node_modules/react-native-ble-manager/ios/CBPeripheral+Extensions.m:52:24 -[CBPeripheral(com_megster_ble_extension) setAdvertisementData:RSSI:]
    node_modules/react-native-ble-manager/ios/BleManager.m:381:5 -[BleManager centralManager:didDiscoverPeripheral:advertisementData:RSSI:]
    Frameworks/CoreBluetooth.framework/CoreBluetooth -[CBCentralManager handlePeripheralDiscovered:]
    Frameworks/CoreBluetooth.framework/CoreBluetooth -[CBCentralManager handleMsg:args:]
    Frameworks/CoreBluetooth.framework/CoreBluetooth ___30-[CBXpcConnection _handleMsg:]_block_invoke
    /usr/lib/system/libdispatch.dylib __dispatch_call_block_and_release
    /usr/lib/system/libdispatch.dylib __dispatch_client_callout
    /usr/lib/system/libdispatch.dylib __dispatch_lane_serial_drain$VARIANT$mp
    /usr/lib/system/libdispatch.dylib __dispatch_lane_invoke$VARIANT$mp
    /usr/lib/system/libdispatch.dylib __dispatch_main_queue_callback_4CF$VARIANT$mp
    Frameworks/CoreFoundation.framework/CoreFoundation ___CFRUNLOOP_IS_SERVICING_THE_MAIN_DISPATCH_QUEUE__
    Frameworks/CoreFoundation.framework/CoreFoundation ___CFRunLoopRun
    Frameworks/CoreFoundation.framework/CoreFoundation _CFRunLoopRunSpecific
    PrivateFrameworks/GraphicsServices.framework/GraphicsServices _GSEventRunModal
    PrivateFrameworks/UIKitCore.framework/UIKitCore _UIApplicationMain
    remotecontrol/main.m:16:12 main
    /usr/lib/system/libdyld.dylib _start
```

marcosinigaglia, 2019-08-23:
The exception it seems to be here:
 https://github.com/innoveit/react-native-ble-manager/blob/57b2152a5c7ad5429db314724990c06862a01fe4/ios/CBPeripheral%2BExtensions.m#L114
Can you try to add a nil check?

folofse, 2019-08-23:
Thank you for your reply.

I can implement a nil check, and publish a new version of my app, it's a bit dodgy because I'm not able to reproduce the problem, it might be that a customer has a faulty BLEPeripheral. But I will try this to start with :)

ianmcgregor, 2020-02-02:
I am seeing this crash too after upgrading to 7.11 from 6.6.9. Platform is iOS 13.3.

Error log:

NSInvalidArgumentException
*** -[__NSPlaceholderDictionary initWithObjects:forKeys:count:]: attempt to insert nil object from objects[2]

dataToArrayBuffer node_modules/react-native-ble-manager/ios/CBPeripheral+Extensions.m:282

-[CBPeripheral(com_megster_ble_extension) serializableAdvertisementData:] node_modules/react-native-ble-manager/ios/CBPeripheral+Extensions.m:115
-[CBPeripheral(com_megster_ble_extension) setAdvertisementData:RSSI:] node_modules/react-native-ble-manager/ios/CBPeripheral+Extensions.m:52
-[BleManager centralManager:didDiscoverPeripheral:advertisementData:RSSI:] node_modules/react-native-ble-manager/ios/BleManager.m:388

Called from:
CoreBluetooth -[CBCentralManager handlePeripheralDiscovered:]

petrusek, 2020-10-01:
I've suddenly started seeing this exception yesterday. Probably some unknown BLE device came online in the neighbourhood or one of my devices around started behaving differently than used to.

**I can confirm the merged fix by @alexkorep worked to eliminate it.** Thank you 👍 

I've used react-native-ble-manager v7.2.0 which did not have this code change yet... after applying the change (to the m file) the issue went away so I've upgraded to v7.3.1 ... everything works now.

closed 2020-10-02 by marcosinigaglia