innoveit/react-native-ble-manager #544: Android scan does not discover any peripherals

Gra55h0pper, 2019-08-22:
### Version

Tell us which versions you are using:

- react-native-ble-manager v6.6.6
- react-native v59.9
- iOS/Android v.8.1.0 on Nexus 6P phone

### Expected behaviour

Would expect to see Bluetooth peripherals to be discovered.

### Actual behaviour

**No peripherals are found whatsoever** on Android Nexus 6P with Android 8.1.0.
(Note that the 'stopScan' event does get triggered after the expected 5 seconds) 

Works fine on iOS and on Android-based Kindle Fire HD10 7th generation. 

### Steps to reproduce

```
import React from 'react'
import { NativeModules, NativeEventEmitter } from 'react-native'
import BleManager from 'react-native-ble-manager'

const eventEmitter = new NativeEventEmitter(NativeModules.BleManager)

export default class App extends React.Component {

  async componentDidMount() {
    
    eventEmitter.addListener(
      'BleManagerDiscoverPeripheral',
      async (args) => {
        console.log("found: " + args.id)
      }
    )

    eventEmitter.addListener(
      'BleManagerStopScan',
      () => {
        console.log("done")
      }
    )

    console.log("started")
    await BleManager.start( { forceLegacy: true } )
    console.log("scanning")
    await BleManager.scan([], 5)
  }

  render() {
    return null
  }
}
```

rayjadore, 2019-08-23:
im having the same issue

marcosinigaglia, 2019-08-23:
You have to debug on your device.

folofse, 2019-08-23:
On android I use the following, sometimes the peripheral seems to be between the connect and disconnect state, 

```
if(Platform.OS === 'android'){
            BleManager.getConnectedPeripherals(servicesToScanFor)
                .then((connectedPeripherals) => {
                    connectedPeripherals.forEach(peripheral =>{
                        this.onDiscoverPeripheral(peripheral);
                    });
                }).catch((error) => {
                    console.log(error);
            });
        }
BleManager.scan(servicesToScanFor, 5, false);
```

Gra55h0pper, 2019-08-23:
@marcosinigaglia Does BleManager have low-level logging I can turn on?

Gra55h0pper, 2019-08-23:
@rayjadore the `scan` method just starts the scan. It doesn't return anything (other than a promise). I would expect your code to return `undefined` on iOS also.

Use the `BleManagerDiscoverPeripheral` **event** to report peripherals found.

Gra55h0pper, 2019-08-23:
@folofse would you mind confirming the Android and react-native-ble-manager versions you use, as well as the Android-phone brand? Are you running on a Nexus 6P by any chance?

folofse, 2019-08-24:
@Gra55h0pper No unfortunately I'm only been able to test on a Google Pixel 2 so far.

Gra55h0pper, 2019-08-25:
**Found the problem!**   😃 

I had missed the Note in the manual stating: **"Android API >= 23 require the ACCESS_COARSE_LOCATION permission to scan for peripherals"**. (and apparently, a lack of permissions does not cause the 'scan' method to trow an exception).

Here's updated code with the fix:

```
import React from 'react'
import { PermissionsAndroid, NativeModules, NativeEventEmitter } from 'react-native'
import BleManager from 'react-native-ble-manager'

const eventEmitter = new NativeEventEmitter(NativeModules.BleManager)

async function getBluetoothScanPermission() {
  const granted = await PermissionsAndroid.request(
    PermissionsAndroid.PERMISSIONS.ACCESS_COARSE_LOCATION,
    {
      title: 'Bluetooth Permission',
      message: 
        'In the next dialogue, Android will ask for permission for this ' +
        'App to access your location. This is needed for being able to ' +
        'use Bluetooth to scan your environment for peripherals.',
      buttonPositive: 'OK'
      },
  )
  if (granted !== PermissionsAndroid.RESULTS.GRANTED) {
    console.log('BleManager.scan will *NOT* detect any peripherals!')
  }
}

export default class App extends React.Component {

  async componentDidMount() {
    
    eventEmitter.addListener(
      'BleManagerDiscoverPeripheral',
      async (args) => {
        console.log("found: " + args.id)
      }
    )

    eventEmitter.addListener(
      'BleManagerStopScan',
      () => {
        console.log("done")
      }
    )

    console.log("started")
    await BleManager.start( { forceLegacy: true } )
    console.log("check location access permission")
    await getBluetoothScanPermission()
    console.log("scanning")
    await BleManager.scan([], 5)
  }

  render() {
    return null
  }
}
```

closed 2019-08-25 by Gra55h0pper