innoveit/react-native-ble-manager #549: Android is disconnecting occasionally with connection error

risingworld777, 2019-09-05:
### Version

Tell us which versions you are using:

- react-native-ble-manager v6.6.8
- react-native v0.59.10
- Android v.9 (Honor V10)

### Actual behaviour

When I run your sample app and do a scan, it is working fine On iOS.
But on android, it is disconnecting occasionally.

`BleManager.connect()` throws exception with `connection error`

It is same issue as https://github.com/innoveit/react-native-ble-manager/issues/270#issue-290131782

risingworld777, 2020-01-08:
@marcosinigaglia  Any update on this issue?

marcosinigaglia, 2020-01-08:
Hi @risingworld777 , I can't reproduce your issue.

EastWooden, 2020-09-11:
Hi @riseworld27, can you finxed this issue? i have geted the same problem on Honnor V10