innoveit/react-native-ble-manager #553: [Android] ClassCastException in LollipopScanManager

OvThAlmin, 2019-09-18:
### Version

- react-native-ble-manager v6.6.8
- react-native v0.59.4
- Android v7.1.2 (SHV40_u, AQUOS sense)

### Expected behaviour

No Exception, or I can understand why App is Crashed by the logs.

### Actual behaviour

I tried to scan some times, Exception occured and my App crashed.
In this case, There are no logs in Chrome console when I use `react-native run-android` .

### Steps to reproduce

1. I kick `BleManager.scan` as follow param:
```
BleManager.scan(
          UUIDs, // string[]
          180,
          false,
          {
            matchMode: 1,
            scanMode: 2,
          })
```
2. When I failed to scan, retry this. 
3. The ClassCastException occured after 5~20 seconds from step2 ↑.
4. App crashed.

### Stack trace and console log
This is in Logcat of Android Studio:

```
2019-09-18 14:26:12.120 21013-21013/{ourApp} E/AndroidRuntime: FATAL EXCEPTION: main
    Process: {ourApp}, PID: 21013
    java.lang.ClassCastException: it.innove.Peripheral cannot be cast to it.innove.LollipopPeripheral
        at it.innove.LollipopScanManager$2$1.run(LollipopScanManager.java:103)
        at android.os.Handler.handleCallback(Handler.java:751)
        at android.os.Handler.dispatchMessage(Handler.java:95)
        at android.os.Looper.loop(Looper.java:154)
        at android.app.ActivityThread.main(ActivityThread.java:6242)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:889)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:779)
```

This is the first time to submit issue in GitHub.
If I need to write more information, kindly tell me.
Thank you.

SunitaMK, 2019-09-23:
Even I am facing same issue.Please help to resolve this.

Device I am using is Samsung edge - Android 7.0

marcosinigaglia, 2019-09-27:
Hi, can you try the release 6.6.10 ?

closed 2019-11-07 by marcosinigaglia

OvThAlmin, 2019-11-11:
For your information,
We updated from 6.6.8 to 6.7.0 last week and checked our apps. Consequently, we confirmed this issue was fixed!
Thank you very much.