innoveit/react-native-ble-manager #560: NSBluetoothAlwaysUsageDescription warning in new iOS 13

xJkit, 2019-11-03:
### Version

Tell us which versions you are using:

- react-native-ble-manager v6.7.0
- react-native v0.61.3
- iOS 13

### Expected behaviour

When the app launches, the BleManager.start function will be called in the componentDidMount lifecycle, and the BleManager should work properly. 

### Actual behaviour

Warning message comes from the Xcode:

```
This app has crashed because it attempted to access privacy-sensitive data without a usage description. The app’s Info.plist must contain an NSBluetoothAlwaysUsageDescription key with a string value explaining to the user how the app uses this data.
```

### Steps to reproduce

1. yarn add react-native-ble-manager
2. Call BleManager.start() function
3. the app crashes and the warning message shows in the Xcode console.

### Stack trace and console log

Hint: it would help a lot if you enable the debugger ("Pause on exceptions" in the "Source" panel of Chrome dev tools) and spot the place where the error is thrown

```
```

marcosinigaglia, 2019-11-05:
Hi, the error is in your app, you need add the new mandatory key `NSBluetoothAlwaysUsageDescription` in your Info.plist

xJkit, 2019-11-05:
Is this better to note this issue on the documentation?

closed 2019-11-07 by marcosinigaglia