innoveit/react-native-ble-manager #564: Library scrambles messages from ble device

SafaTinaztepe, 2019-11-14:
### Version

Tell us which versions you are using:

- react-native-ble-manager v6.7.0
- react-native v0.61.3
- iOS/Android v.13.2.2 on iPhone 6 and iPhone 11

### Expected behaviour

I am using the [Adafruit Bluefruit LE device](https://www.adafruit.com/product/2479) with Arduino to send sensor data to a mobile RN application using ble-manager.  I attempt to use the `BleManager.startNotification` method to read notifications from ble device UART serial output and write it to a file or a Text component.  The expected behavior is to read messages coming in from the device and to preserve the message as it is printed in the Arduino console.  The expected message is also correctly received in the Bluefruit Connect OSX app and the Bluefruit Connect iOS app.

### Actual behaviour

When I attempt to retrieve this message using the ble-manager library, the notifications I receive with `startNotification` returns a scrambled version of the message, drops characters, and receives messages out of order.  This behavior is observed in iPhone 6 and iPhone 11 on the latest iOS version.

### Steps to reproduce

1. Connect to device using `device.connect(UUID)`
2. Discover services and characteristics on connection with device
3. Execute `startNotification`
4. on update value, write value to local file, upload to cloud and view 

### Stack trace and console log

let messageIndex *i*,
     time *t*,
     sensorValue1 *y1*, 
     sensorValue2 *y2*;

#### Example output observed in both sender Arduino console and receiver Bluefruit Connect
{i:195,t:12241,y1:257,y2:254}
{i:196,t:12276,y1:257,y2:254}
{i:197,t:12311,y1:257,y2:254}
{i:198,t:12346,y1:257,y2:254}
{i:199,t:12380,y1:256,y2:253}
{i:200,t:12414,y1:257,y2:253}
{i:201,t:12449,y1:257,y2:254}
{i:202,t:12484,y1:257,y2:254}
{i:203,t:12519,y1:256,y2:254}
{i:204,t:12554,y1:257,y2:254}
{i:205,t:12588,y1:257,y2:254}
{i:206,t:12622,y1:257,y2:254}
{i:207,t:12657,y1:257,y2:254}

#### Example output received in `react-native-ble-manager`

531}
{i:4127,t:153105:4129,t:154045,y110{i:4128,t:1540085771,y1:254,y2:251}44y1:254,y2:251}
{58254,y2:251}
{i:44930,t:154083,y1:2552,y2:251}
{i:413144t:154120,y1:254,y50:251}
{i:4132,t:4954157,y1:254,y2:2531}
{i:4133,t:1544993,y1:254,y2:251}10{i:4134,t:15423044y1:254,y2:251}
{105:4135,t:154267,y1:5054,y2:251}
{i:52136,t:154305,y1:2552,y2:251}
{i:4131212:251}

### Discussion

- What do you think is going on here?
- Is there another recommended way to read a stream from a device?  Is it better to chain `device.read` calls than to use `device.startNotification`?

marcosinigaglia, 2019-11-15:
Hi, maybe the writing is too fast for the RN app. I don't know how the device is working so I don't know if is better to use `read` instead the `startNotification`.

SafaTinaztepe, 2019-11-15:
That is a good point.  I'll introduce rate throttling in Arduino.  This phenomenon is likely not an issue with the library, so I'll close the ticket.  Thank you for your suggestion.

closed 2019-11-15 by SafaTinaztepe