innoveit/react-native-ble-manager #565: In iOS 13 Bluetooth pair permission popup appear multiple times, block the UI

pradeep11, 2019-11-15:
### Version

Tell us which versions you are using:

- react-native-ble-manager v 6.7.0
- react-native v .60
- iOS/Android iOS 13?

### Expected behaviour
Bluetooth pairing alert should be appear only once

### Actual behaviour
Bluetooth pairing alert should be appear multiple time and block UI in iOS 13,
Working fine with the iOS 12

Tell us what happens instead

### Steps to reproduce

1. Integrate Library in react native project
2. When try to initialise the bluetooth, the pairing alert appear.
3. Click on pair option.
4 Again alert appears and it continuously apearing after that.

Tested in iOS 13, but same flow is working fine for iOS 12

marcosinigaglia, 2019-11-26:
Hi, I can't debug this, you have to try yourself.

moltedopablo, 2021-11-29:
Any news on this? I have the same issue