innoveit/react-native-ble-manager #567: Data collision in two write characteristics

erdouzun, 2019-12-03:
### Version

Tell us which versions you are using:

- react-native-ble-manager v6.6.1
- react-native v0.58.3
- iOS/Android IOS 13.1.2

### Expected behaviour

I write a data into Write Characteristic 1:

{"data":"S3xPyEy8TqqmLK7tcq8Uo22WVtoq3bBr0o4Rw7YDb5CJEKWC7XCn74+SIbmh30o2mf6zEgzeERYAuGfjjSGNbslb0fguL8yyT/uJTwDeKEig5JL2hcBdno84H1m36+NUmx5KTDJ32/dkhJ/36sY9+2IvoBomi/bydyZdYCQgcaC2hDzXSL7bqbBiLhYhGJ3uxQSWDjc+LxXo0nVog7HTds/ZRvRP6Ph3LPppDxIdWn4CZpzoVBWeEcdHAJb6v7XmBcmScmOBPrlTjtXRnFJ679v8TPjZpuzIPI7vbmqdSRlNFD/ScEQAcNJ59Fk+1wrNT0qa8KLTLy9Wh5sfw5xjslnDHbJH7zNVeaPu/VIaQGmo1YKMX8N/lyvCCB6doVFzCruf43JDLQQDO0WypIN/Igtdt/7PofT+mMVjGnz36IHpIRCnPwF34CKxKx7uCSZZ1ZyKneXS5H92O2uOYs65ro1Ej4XqAzV2DvHhXbM/FFlBPH1YIzJqneovKQI5wFfr5itX/F6IMV0fQ19cVpBUyKkNUyCJtzrLV+uD4mylALYsjhDHx5JGpUBg2x9N6mobzgAGJYg5sV75LwRQ5dqO5jRQbPlwTXbPSXuJ00Y+2uM2si8/ch31tIbPxvAKOTJDB4rLKJl2UNedlZ6fHjwNgdWBEKFAkBKXzQMxjM+1riD9l06pzmPJQR5JlDoNm+1pUEbYth4XJqAjX5eY6t5429ArApdKuVwUbeEqX/1wBkZLeyydhlFVg1qmed9kL0raU62IFJWg2hXR3H9Oz234Mm077V/T4r4Fn2/+6JsCrRdWSPpuJnKSErEFHuvgSav14tBY2s/u53e42yuSjSFbcAVhBJDpMQk4L3suwjI8YDyxoieqmOIK4MyN7q6arWaxS2WsLtVFdger+T108ReSS/oOY/iyVKxSclspNySu1XrH1BYhc138D2I+JkonNjXz2YdWVHPRD60TOAVngS7wpHtvRHAuIMCHWVo90i+7NZ3m7uRYdRqge0CSLXvR0Ky8GIcZaa1HoWhhtRHC02NtM0QBotP8GQeqSpfyybvjOP1bHWxIBrRo/uNBcpwCuuxh5Peef8+/TVzXUjnh1II5porYFWBX3YlyA190S3wEpqwu4Rh1R64WsoQ9z7SZDRkSIC8LNMtb/Xh1Pn5I/rTLT6ZDY9cbbPmQ4J5jfOPgtP3kYRt9yDop4omf4zP6UxRl+sChqj+B/jtDcI9eE77nhQ=="}

And after I write another data to Write Characteristic 2:

{"data":"9a+9WjZE67xYmhY2Q9Xulj48W4mEv0EREJ95TUgKUD+O/nJMLlRrc5+6N32nf+YSSGniqw89gb3MoPnIsw6OWfEMD3Pl7C8npFKiJ+7NZg36raAPq1PE9iiNRpkbv8K88wmoJbL5gpVIGfTwXCN9Vir7JLNCj8QoMU96WcdALoc="}

We expect The BLE device to read successfully in different characs.

### Actual behaviour

BLE device read collided data which is concatted with Characteristic 2 in Characteristic 1. Not a full concatted but it is not the format that we expected.

We examine the sent data logs in ble manager, everything is fine. On the other side, ble device logs corrupted/collided data.

Can you give any advise about this problem? I hope I can clarified the problem well.

marcosinigaglia, 2019-12-03:
In my experience is always better to add a delay between different writes.

erdouzun, 2019-12-04:
Thanks for your help.

We also think about differing the period of data sending; but in some cases we cannot know the exact moment of writings of these two characteristics.

What do you think about the main reason of this situation? Is it a general ble/bluetooth problem or ble-manager library?

marcosinigaglia, 2019-12-04:
Is a general bluetooth "problem", in my projects I use a queue system to avoid concurrent writes.

dbarth, 2020-09-09:
I think this PR should help: https://github.com/innoveit/react-native-ble-manager/pull/660

closed 2022-08-12 by marcosinigaglia