innoveit/react-native-ble-manager #568: Write Image or Qr Code

JuanCamilo0809, 2019-12-04:
Hi,
 
I want to write a Qr Code or any image in a printer, i can write text ok, but i can´t do it for image or qr code.

Somebody know how do it?

thanks.

FazilMuhammed, 2020-10-12:
@JuanCamilo0809 Hello Brother. did you find the solution.i can also write text but i want to write image in printer any possible please help me.

JuanCamilo0809, 2020-10-13:
@FazilMuhammed Hello,

I can not do it. I use https://github.com/januslo/react-native-bluetooth-escpos-printer library and work fine.

FazilMuhammed, 2020-10-14:
@JuanCamilo0809 Which Platform You running IOS or Android..?

FazilMuhammed, 2020-10-14:
i have trouble with IOS can help me ? which is your react native version what are steps you done.?

JuanCamilo0809, 2020-10-14:
I use
"react": "16.9.0",
"react-native": "0.61.2",
"react-native-bluetooth-escpos-printer": "git+https://github.com/januslo/react-native-bluetooth-escpos-printer.git",

and its working fine in Android and IOS

FazilMuhammed, 2020-10-17:
@JuanCamilo0809 Now i followed this Library react-native-bluetooth-escpos-printer i have some doubts.while i'm connect to my thermal printer 

here is my code to connect to bluetooth thermal device
  
 _renderRow(rows) {
        let items = [];
        for (let i in rows) {
            let row = rows[i];
            if (row.address) {
                items.push(
                    <TouchableOpacity key={new Date().getTime() + i} style={styles.wtf} onPress={() => {
                        this.setState({
                            loading: true
                        });
                        console.log(row.address, "Address")
                        BluetoothManager.connect(row.address)
                            .then((s) => {
                                console.log(s, "S")
                                this.setState({
                                    loading: false,
                                    boundAddress: row.address,
                                    name: row.name || "UNKNOWN"
                                })
                            }, (e) => {
                                this.setState({
                                    loading: false
                                })
                                alert(e);
                            })

                    }}><Text style={styles.name}>{row.name || "UNKNOWN"}</Text><Text
                        style={styles.address}>{row.address}</Text></TouchableOpacity>
                );
            }
        }
        return items;
    }

when i debug getting undefined inside connecting my iPhone device to printer and after this function call printer is connected looking this i happy but i'm confused why this undefined happen and also send the command to printer 

here is command

  printSomething = async () => {
        //with normal text, mine is 58mm mini thermal printer and has 32char limitation
        console.log("Hello Fazil")
        await BluetoothEscposPrinter.printText("------------------------------\r\n", {});
        await BluetoothEscposPrinter.printText("This is another text", {});
        await BluetoothEscposPrinter.printText("------------------------------\r\n", {});
        //with columns, split to 4 columns of 8 char each, need to specify alignment for each column and text for each column
        await BluetoothEscposPrinter.printColumn([8, 8, 8, 8], [BluetoothEscposPrinter.ALIGN.RIGHT, BluetoothEscposPrinter.ALIGN.RIGHT, BluetoothEscposPrinter.ALIGN.RIGHT, BluetoothEscposPrinter.ALIGN.RIGHT],
            ["column1", "column2", "column3", "column4"], {});
        await BluetoothEscposPrinter.printColumn([8, 8, 8, 8], [BluetoothEscposPrinter.ALIGN.RIGHT, BluetoothEscposPrinter.ALIGN.RIGHT, BluetoothEscposPrinter.ALIGN.RIGHT, BluetoothEscposPrinter.ALIGN.RIGHT],
            ["1", "PRODUCTA", "2", "$2.00"], {});
        await BluetoothEscposPrinter.printColumn([8, 8, 8, 8], [BluetoothEscposPrinter.ALIGN.RIGHT, BluetoothEscposPrinter.ALIGN.RIGHT, BluetoothEscposPrinter.ALIGN.RIGHT, BluetoothEscposPrinter.ALIGN.RIGHT],
            ["2", "PRODUCTB", "3", "$1.00"], {});
    }

this function also calls inside componentDidMount() too the result nothing happening...
 

can you share your code to me please to check. its will be great appreciation
i will share my email id here fazilmuhammed299@gmail.com waiting your response.

JuanCamilo0809, 2020-10-17:
@FazilMuhammed Dont worry, I help you.

closed 2020-10-17 by JuanCamilo0809

FazilMuhammed, 2020-10-18:
@JuanCamilo0809  Thanks for your response...

_testPrint = async () => {
        const Printerid = await AsyncStorage.getItem('PrinterId');

        console.log(Printerid, "printerID")

        if (Printerid == 'null' || Printerid == null || Printerid == '') {
            Alert.alert('No has configurado impresoras previamente');
            return;
        }

        ///Primero se intenta conectar a la ultima impresora que tuvo emparejada
        const connect = await BluetoothManager.connect('3D7FB052-4854-C524-FCC8-1233061B6D32');

        console.log(connect, "Connect")

        const result = await BluetoothEscposPrinter.printText('Prueba App Movil PosGold \n\n\n\n', {});

        console.log(result, "Result")
    }

when i running i try to console the connect i got undefined value i dont how its happen and my printer does not happen anything when send the text to printer..