innoveit/react-native-ble-manager #572: 133 error with Samsung S10 with Android 10

AnaRockbell, 2020-01-10:
### Version

Tell us which versions you are using:

- react-native-ble-manager v6.6.8
- react-native v0.60.5
- Android v.10 on Samsung Galaxy S10 & Note 10

### Expected behaviour

I am sending wifi credentials over bluetooth to a device. With every device other than the listed ones above, when I call BLEManager.java's .write() to send over the wifi id & password it works as intended. 

### Actual behaviour

However, on the devices noted above error 133 is thrown every time.

### Steps to reproduce

1. Call BleManager.write with 5 inputs (no callback) on one of the specific above devices.
2.
3.

### Stack trace and console log

2020-01-10 14:10:24.777 1987-2124/? D/BtGatt.GattService: onDisconnected() - clientIf=12, connId=268status=8, address=11:8
2020-01-10 14:10:24.778 18665-19408/com.iqua E/ReactNativeBleManager: Error onCharacteristicWrite:133
2020-01-10 14:10:24.779 18665-19408/com.iqua D/BluetoothGatt: onClientConnectionState() - status=8 clientIf=12 device=[REDACTED]
2020-01-10 14:10:24.782 18665-19408/com.iqua D/ReactNativeBleManager: onConnectionStateChange to 0 on peripheral: [REDACTED] with status8
2020-01-10 14:10:24.782 18665-19408/com.iqua D/BluetoothGatt: cancelOpen() - device: [REDACTED]
2020-01-10 14:10:24.786 1987-2124/? D/BluetoothUtils: getBtEnabledContainers(): btContainers = []
2020-01-10 14:10:24.787 1987-2124/? D/BluetoothDataManager: BluetoothDataManager :: notifyDisconnectReason() is called!!!
2020-01-10 14:10:24.787 1987-2124/? W/bt_btif: BTA_DM_BUSY_LEVEL_EVT :: p_data->busy_level.level_flags : 0x0, spen_connected : 1
2020-01-10 14:10:24.791 1987-1987/? D/BluetoothAdapterService: getAdapterService() - returning com.android.bluetooth.btservice.AdapterService@39071a1
2020-01-10 14:10:24.793 4418-4418/? I/oneconnect[1.7.41-25_06][CORE]: BluetoothHelper.handleActionConnectionStateChanged -  [action]android.bluetooth.device.action.ACL_DISCONNECTED [connState]0[address] - ($)
2020-01-10 14:10:24.797 1987-1987/? D/PanService: onReceive : android.bluetooth.device.action.ACL_DISCONNECTED
2020-01-10 14:10:24.797 1987-1987/? D/BluetoothPbapService: action: android.bluetooth.device.action.ACL_DISCONNECTED

marcosinigaglia, 2020-01-15:
Hi, I don't have your device so you have to debug yourself.

kulyk, 2020-01-15:
Hi @marcosinigaglia,

I face the same issue @AnaRockbell has. In my app, the scan is stopped before writes

I've tried to reproduce the issue on different phones with different OS versions and all works fine (Google Pixel, One Plus, Huawei with Android 10, Samsung Galaxy S8 with Android 9, One Plus 5 with Android 8, Xiaomi with Android 7).

We have a few dozens of bug reports from our users with the latest Samsungs. So the problem definitely reproduces only with Samsung Galaxy (Note) 10 and Android 10

Is there any hypothetical fix or problem cause you see?

kulyk, 2020-01-15:
After quick research I've found out the latest Samsungs can have a wide range of problems with Bluetooth devices. There are bug reports with Bluetooth headphones, cars, etc. You can read more about it [here](https://piunikaweb.com/2019/04/01/samsung-galaxy-s10-bluetooth-problems-issues-reported-with-galaxy-buds-and-cars-like-bmw-mini/)

Samsung has even published a [post about problems with cars' Bluetooth services](https://www.samsung.com/sg/support/mobile-devices/what-to-do-when-samsung-galaxy-s10-series-cannot-bluetooth-pair-with-bmw-mini-car-audio/).

We'll try to test a few recommended solutions:

1. [Resetting network settings](https://www.samsung.com/sg/support/mobile-devices/what-to-do-when-samsung-galaxy-s10-series-cannot-bluetooth-pair-with-bmw-mini-car-audio/)
2. [Turning off battery optimization for an app](https://us.community.samsung.com/t5/Galaxy-S10/Galaxy-S10-Plus-Bluetooth-Disconnecting/td-p/472781)

I'll get back here later with the results

marcosinigaglia, 2020-01-15:
In my experience the bluetooth in Android it's a mess. I hope in the next days to release a version with the support for AndroidX, but I don't think that solve the problems.

kulyk, 2020-01-15:
Yeah, it definitely is 😔 Looking forward to Android X release and many thanks for your work!

marcosinigaglia, 2020-01-16:
Hi @kulyk , can you try the new release?

kulyk, 2020-01-16:
No luck with the approaches above

kulyk, 2020-01-20:
Hi @marcosinigaglia, the issue with Samsungs still reproduces with `ble-manager` 7.0.0 😞

kulyk, 2020-01-24:
Hi @marcosinigaglia @AnaRockbell 

We've got Samsung Galaxy S10 for debugging, would like to share a few things we discovered

We are able to fix characteristic write if we remove the `requestMTU` call. Data packages are transmitted by 18 bytes chunks, writes are extremely slow, but at least it works.

Also if the app requests MTU to be 23 or lower, the `requestMTU` promise never resolves

kulyk, 2020-01-24:
OMG, the issue reproduces even with [NRF Connect app](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=en_US)

kulyk, 2020-01-28:
@AnaRockbell hi, what's the chip you are using on a peripheral device?

kulyk, 2020-01-29:
@AnaRockbell hi,

Somebody had faced the same issue we have and created a bug report ticket in the Samsung Developers forum. We're going to comment and add some info to the report, so Samsung notices the issue

Please join us: https://forum.developer.samsung.com/t/samsung-android-10-ble-connectivity-regression/509

kulyk, 2020-01-29:
@marcosinigaglia hi. I guess this issue can be closed. Looks like the problem is on Samsung side

marcosinigaglia, 2020-01-29:
Thanks @kulyk

closed 2020-01-29 by marcosinigaglia