innoveit/react-native-ble-manager #573: IOS 13 BleManagerDiscoverPeripheral sometimes can't receive ble device, sometimes can

zhao812, 2020-01-15:
### Version
Tell us which versions you are using:

- react-native-ble-manager v 6.7.0
- react-native v0.61.5
- iOS/Android IOS 13

### Expected behaviour
BleManagerDiscoverPeripheral sometimes can't receive ble device, sometimes can

### Actual behaviour

Tell us what happens instead

### Steps to reproduce

1.
2.
3.

### Stack trace and console log

Hint: it would help a lot if you enable the debugger ("Pause on exceptions" in the "Source" panel of Chrome dev tools) and spot the place where the error is thrown
this._handlerDiscover = BleManagerEmitter.addListener('BleManagerDiscoverPeripheral', this._handleDiscoverPeripheral);
BleManager.start({ showAlert: false }).then(() => {
				console.log('Init the module success.');
				BleManager.scan([], 10, true).then(() => {
					// Success code
					this.setState({ isScan: true })
					console.log('Scan started');
				});
			}).catch(error => {
				console.log('Init the module fail.');
			});

marcosinigaglia, 2020-01-15:
Hi, I don't thinks is a library problem.

zhao812, 2020-01-15:
Thank you!
Because I call function start in componentDidMount. 
call this function start in button to resolve it.

closed 2020-01-15 by zhao812