innoveit/react-native-ble-manager #580: [Question] Background scanning supported?

gluonplasma, 2020-01-29:
### Version

- react-native-ble-manager v7.0.0
- react-native v0.61.5
- iOS/Android v13.3 iPhone X

I've been trying to get background BLE scanning to work (https://developer.apple.com/library/archive/documentation/NetworkingInternetWeb/Conceptual/CoreBluetooth_concepts/CoreBluetoothBackgroundProcessingForIOSApps/PerformingTasksWhileYourAppIsInTheBackground.html) 

What I've tried so far:
-enabled background scanning in the plist.info file (UIBackgroundModes > bluetooth-central)
-in componentDidMount() of the example modified `this.handlerDiscover` to something more basic
```
    let args = {
      id: '54EEDAE3-A28E-ACB4-E99D-38CAF078A13B',
      name: 'BLE Device 1'
    }

    this.handlerDiscover = bleManagerEmitter.addListener('BleManagerDiscoverPeripheral', (args) => {
      console.log('################ BLE DEVICE 1 FOUND ##################');
    } );;
```

This partially works when the app is in the foreground in that it prints the "BLE DEVICE 1 FOUND.." every time a device is discovered regardless if it is or isn't the one I'm looking for.  When the app is in the background it stops scanning altogether.  

I'm at a point where I think either background scanning is not supported by this library, or iOS isn't picking up that device in the background because it was already found in the foreground of my app (according to the first answer of https://stackoverflow.com/questions/32140863/ios-diddiscoverperipheral-not-called-in-background-mode) 

So my question : is background BLE scanning in iOS supported by BleManager?

marcosinigaglia, 2020-02-25:
Hi, background scanning is not supported.

MD3XTER, 2020-03-02:
> Hi, background scanning is not supported.

If it's not supported, then how am I being able to scan and connect in the background? I just followed this: https://stackoverflow.com/a/39825460/5998562

marcosinigaglia, 2020-03-03:
Hi @MD3XTER , you need a custom module that handle the background scanning natively.

gluonplasma, 2020-03-04:
ok, thanks

closed 2020-03-04 by gluonplasma

hchaudhary1, 2020-09-07:
@gluonplasma were you able to get it working using a custom module?