innoveit/react-native-ble-manager #582: BleManager Scan Fails

renatoramossilva, 2020-02-04:
### Version

Tell us which versions you are using:

- react-native v0.61.5
- iOS/Android v.9.1.0

Hi,

I am trying to use the ble-manager library. My first test was run the example and I can see the devices connected in my android using the getConnectedPeripherals() method.

Now, I wanna to list all the idle bluetooth devices.. for to do it I am using the scan method.

BleManager.scan([], 10, true).then(results => {
        console.log('Scanning...');
        this.setState({scanning:true});
      });

The behavior in my android 9.1.0 is an error when I called the scan method.

//isNull - ReadableNativeMap.java:102
//scan - LollipopScanManager.java:48
//scan - BleManager.java:176

I think the error is here: 
		if (scanManager != null) scanManager.scan(serviceUUIDs, scanSeconds, options, callback);

But I am not sure about this issue.

Please, can anyone help me with this, I dont know if there is any compatible problem or something like this.

Tks
```
```

Gueoff, 2020-02-25:
Hello!
I have the same problem as you, did you find a solution?

sophana, 2020-05-26:
same for me. in the example app, scanning shows the same error.
(I had to change the manifest to make it run)
running android 10 
xiaomi mi9 lite
miui 11

viren-vii, 2021-04-18:
Have you resolved it? If yes please tell how :)

closed 2023-12-04 by marcosinigaglia