innoveit/react-native-ble-manager #583: Multiple devices

srossi3, 2020-02-05:
Hi everyone! 
Would it be possible to connect multiple devices to an app using this library?
In particular, I would need to read controls from one bluetooth device, process them in the application and send new controls to a different device. 
Thank you, 
Silvia

marcosinigaglia, 2020-02-06:
Hi @srossi3 , I think is possible, the only limitation is that you can't read or write simultaneously.

glenne, 2023-06-09:
I've been using the library with multiple devices for several years not.  However, I just upgraded from 7.4.1 to 10.0.2 and on iOS I cannot do two calls to `await BleManager.retrieveServices(id)` simultaneously with different ids.   The 1st one never returns.  Serializing each await works properly.  Is this part of 'do not read or write simultaneously' or is it a bug I should investigate?