innoveit/react-native-ble-manager #585: Make isConnectable working for API 26 and up

juliansteenbakker, 2020-02-24:
Right now the isConnectable flag is set to true for any device found in Android.
`advertising.putBoolean("isConnectable", true);`

[However, since API 26 (Android 8) it is possible to call isConnectable on the scanresult, which returns a boolean](https://developer.android.com/reference/android/bluetooth/le/ScanResult.html#isConnectable())

Is it possible to check if API version is 26 or up, and pass the value from the scanrecord?

marcosinigaglia, 2020-02-25:
Hi @juliansteenbakker , yes a PR is welcome

juliansteenbakker, 2020-02-25:
Hello @marcosinigaglia , See https://github.com/innoveit/react-native-ble-manager/pull/589

marcosinigaglia, 2020-02-26:
Hi @juliansteenbakker , the new release have the PR

closed 2020-02-26 by marcosinigaglia