innoveit/react-native-ble-manager #586: Android fail scan/connect

Gueoff, 2020-02-24:
Hello!

I can't use this library anymore for a week because it doesn't work on android devices. It works fine on ios.
I don't scan anything or connect to any peripheral.
So I have recreated a new project with this lib only and the same thing happened.

Is there a step that I forgot ?

### Version

- react-native-ble-manager v6.7.0
- react-native v0.61.4
- iOS/Android v.8.0

### Expected behaviour

The scan should detect peripherals

### Actual behaviour

No device is scanned on android

### Steps to reproduce

1. Create new RN project
2. Install react-native-ble-manager library with yarn (yarn add react-native-ble-manager)
3. Copy/paste /example/App.js in my App.js file on my project
4. Add permissions in AndroidManifest.xml
5. Launch app
6. Scan peripherals (always empty array)

marcosinigaglia, 2020-02-25:
Check the permission of the app, probably is a phone problem.

Gueoff, 2020-02-25:
I have tested the same app on 2 different devices :
- WIKO W_C800 / android 8.0.0
- HONOR 10 COL-L29 / android 9.1.0

The scan works only on HONOR, permissions are ON for the two devices.
Thanks for your response ;)

closed 2020-02-25 by Gueoff

stevesuyao, 2020-02-27:
Confirmed. Android 8 failed to scan.