innoveit/react-native-ble-manager #588: Automatic disconnect with iOS 13 and Nordic BLE devices

oscar-gallog, 2020-02-24:
### Version

Tell us which versions you are using:

- react-native-ble-manager v6.6.8
- react-native v0.61.0
- iOS v13.3

### Expected behavior
it should search for devices and then connect
Tell us what should happen

### Actual behavior
It should search for BLE devices and then connect to the BLE device.

Tell us what happens instead
It connects for a second, and then it disconnects, the message is:
`[B5A226CB-12A3-53FE-FE1C-470D2489453F] Disconnected`
### Steps to reproduce

1. Install the library.
2. run command .start.
3. connect to the BLE device.
4. less than half a second after you get disconnected.

### Stack trace and console log

Hint: it would help a lot if you enable the debugger ("Pause on exceptions" in the "Source" panel of Chrome dev tools) and spot the place where the error is thrown

```
[B5A226CB-12A3-53FE-FE1C-470D2489453F] Disconnected
--
```

This only happens on iOS 13 >.
And with nordic BLE devices, but the weirdest thing is... this works properly on iPad Pro and regular iPad, but on iPhone iOS it disconnects you every time, we try with 6 different iPhones and we are having the same issue.

marcosinigaglia, 2020-02-25:
Hi @oscar-gallog , did you try to use other app like https://apps.apple.com/us/app/lightblue/id557428110 ?

oscar-gallog, 2020-03-01:
I installed the latest update @marcosinigaglia  and is connecting, but sadly the method retrieveServices is not working for me, at all, it just stops right there.

The device is now connected, and the device stopped searching for connections, so after I run `.connect`
I try to get the services, and nothing happens, literally, it just stops, no error, no nothing. I tried to add setTimeouts, from 200ms to 4000ms and nothing works, any idea why?

```
let peripheralInfo: PeripheralInfo;
    try {
      peripheralInfo = await BlaManager.retrieveServices(deviceId);
     console.log('Services info', peripheralInfo);
    } catch (ex) {
      console.log('Error whjle retrieving services', ex);
    }
```

oscar-gallog, 2020-03-02:
@marcosinigaglia Light blue is unable to connect to the Nordic BLE device, this library is connecting, but fails in the retrieving service function

marcosinigaglia, 2020-03-03:
I think is a device problem.

closed 2023-12-04 by marcosinigaglia