innoveit/react-native-ble-manager #590: [iOS] Handling pairing / didUpdateNotificationStateForCharacteristic error

CptFabulouso, 2020-02-26:
### Version

Tell us which versions you are using:

- react-native-ble-manager v6.7.0
- react-native v0.59.10
- iOS v.13

### Expected behaviour

Be able to listen for errors in `didUpdateNotificationStateForCharacteristic` during registering for notifications and the function `BleManager.startNotification()` should resolve or reject.
I would expect the callback happens here https://github.com/innoveit/react-native-ble-manager/blob/de7b7c9c6ff4c426388e8d867038b8038ccf82c6/ios/BleManager.m#L109

### Actual behaviour

The error is only logged to xCode (sometimes no error, see below), function `BleManager.startNotification()` is not resolved or rejected and the app awaits forever.

### Steps to reproduce
It's a pairing issue, it happens when I remove the paring device in one of the devices, with slightly different behavior.
note I do the necessary steps to be able to start notification every time.

**First case**
1. pair with BT device (everything works at this point)
2. remove pairing info from the phone
3. pair with BT device again (the behavior above happens)

In the function "didUpdateNotificationStateForCharacteristic" I get error "Encryption is insufficient", or "Authentication is insufficient". That's ok, but I am not notified nor the startNotification ever resolves or gets rejected. I have to set timeout in js and reject it manually.

**Second case**
1. pair with BT device (everything works at this point)
2. remove pairing info from the BT device
3. pair with BT device again (the behavior above happens)

In the function "didUpdateNotificationStateForCharacteristic" there **sometime** was an error "The specified UUID is not allowed for this operation", but mostly nothing happens, no error, the startNofication listener stalls.
 
I don't know if there is an pairing issue in the iOS system itself, or in the BT device, but when they can't pair, there is no error on js side.

marcosinigaglia, 2020-02-26:
Hi @CptFabulouso , can you make a PR?

CptFabulouso, 2020-02-27:
I may look into it, but I am not sure how at the moment, I am not very experienced with native code. Someone with better native experience might do it better and faster.

marcosinigaglia, 2020-02-28:
Hi @CptFabulouso , can you try the new release? (7.1.4)

CptFabulouso, 2020-03-05:
Same behavior

amor87, 2020-07-21:
@CptFabulouso
Quick question man, due to it seems you made it work to the connecting part, I've been trying to unsuccessfully

After Scanning for devices, I detect in close range my Bluetooth headphones, and I see it on the peripheral detected list, so then when I try to connect to them I can see the log in Xcode that is trying to connect to the device 
`Connecting to peripheral with UUID : `
but then it does nothing else.

I tried to connect the device with my Mac and it worked, but I was not able to it with my headphones and external keyboard... not sure if it's because the property isConnectable returns as false meaning that I can't connect to those devices, which is weird because if I can pair them through the device settings.

Did you include any special code on Objective-C/Swift?

CptFabulouso, 2020-07-26:
@amor87 I didn't modify the native code, I only made some workaround with timeout on js side to prevent never resolving request. Not sure where the cause of your issue is.

closed 2023-12-11 by marcosinigaglia