innoveit/react-native-ble-manager #593: Design recommendation please

anwarhamr, 2020-02-27:
### Version
7.1.1

Tell us which versions you are using:

- react-native-ble-manager v7.1.1
- react-native v61.5
- iOS/Android Android 6+

### Expected behaviour
I have a custom ble device that sends data at interval ~6 seconds.  It sends 256 messages via characteristic updated.  My app is crashing after ~3hr interval but I don't know if the source is this module so more on this below if you care.  I think sending each value across the bridge is a performance hit.  I would like to either add functionality to your code or make a perm. fork.  

I'm looking for some input 1) whether or not my idea would be a good feature for your project. 2) an idea on architecture even if you're not going to include it because I'm new to java but have much experience w/ c# and programming in general.  

Our currently solution requires only Android support, hence my thoughts below cover the Java side of things.  Our device is sending bursts of data at regular interval.  I would like to buffer that data on the java/ios side and send updates on user specified service/characteristic at a time interval or a number of items buffered.  How would you do something like this?  I have used Rx and there are things like bufferTime and bufferCount; i'm not yet sure if JavaRx would work on android.  The other option would be to use Timers.  I'll obviously have to look into some things but one initial thought is would I be dealing w/ threads and do I need to use concurrent structures etc.  Any thoughts would be helpful.  Thanks for your time.

Tell us what should happen

### Actual behaviour

### Stack trace and console log

Hint: it would help a lot if you enable the debugger ("Pause on exceptions" in the "Source" panel of Chrome dev tools) and spot the place where the error is thrown

```

02-25 03:23:24.862 17936 18208 D ReactNativeBleManager: Read: 2A1100EC25C7F5A05916F71E2A9FA7F8D1D25BD5 from peripheral: 80:EA:CA:70:01:14
02-25 03:23:24.864 17936 18208 D ReactNativeBleManager: Read: 2A1100EDE635FA9768399A1866D535A5BD8A6709 from peripheral: 80:EA:CA:70:01:14

02-25 03:23:24.898 17936 18208 W com.auraer: JNI critical lock held for 21.477ms on Thread[41,tid=18208,Runnable,Thread*=0x71a31e9000,peer=0x12c41d18,"Binder:17936_4"]
^^^^^

02-25 03:23:24.899 17936 18208 D ReactNativeBleManager: Read: 2A1100EEFE367657E519787B261A3267809A35BD from peripheral: 80:EA:CA:70:01:14
02-25 03:23:24.902 17936 18208 D ReactNativeBleManager: Read: 2A1100EF6068E169BA7D949E1611515F5DD92673 from peripheral: 80:EA:CA:70:01:14
02-25 03:23:24.905 17936 18239 D ReactNativeBleManager: Read: 2A1100F0986106D96058DA88326BA16467799259 from peripheral: 80:EA:CA:70:01:142A1100FD95A483662A2A5A7A5845817472A379F2 from peripheral: 80:EA:CA:70:01:14
02-25 03:23:24.959 17936 18239 D ReactNativeBleManager: Read: 2A1100FE838DDBC5A562665F7992969ADFC63D61 from peripheral: 80:EA:CA:70:01:14
02-25 03:23:24.961 17936 18239 D ReactNativeBleManager: Read: 2A1100FF601C48DE809F24274168559927D25B00 from peripheral: 80:EA:CA:70:01:14

02-25 03:23:25.934 17936 17960 D NetworkManagementSocketTagger: tagSocket(58) with statsTag=0xffffffff, statsUid=-1
02-25 03:23:26.464 17936 17971 I HermesGC: OOM: [HermesRuntime] reason =                                                                                                                             Max heap size was exceeded (1 from category:                               vm_allocate_category), numCollections = 13357, heapSize = 538722304, allocated = 530771640, va = 549453824.
02-25 03:23:26.464 17936 17971 I HermesGC: NCGen OOM: #segments = 131, !materialize = 0, maxHeapSize = 538722304
02-25 03:23:26.468 17936 17971 W com.auraer: 0xebadde09 skipped times: 0
02-25 03:23:26.469 17936 17971 F libc    : Fatal signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x0 in tid 17971 (mqt_js), pid 17936 (com.auraer)

02-25 03:23:26.589 17936 18209 D ReactNativeBleManager: Read: 841F0000842368C61C77DB56A5CA7CE228EA3389 from peripheral: 80:EA:CA:70:01:13
02-25 03:23:26.616 17936 18209 D ReactNativeBleManager: Read: 841F00070725AB6C9D782E348E2257EDBA6D9C37 from peripheral: 80:EA:CA:70:01:13
02-25 03:23:26.618 17936 17949 D ReactNativeBleManager: Read: 841F0008D6248FA3485DDB6B1BB7220D8CA2985D from peripheral: 80:EA:CA:70:01:13
02-25 03:23:26.620 17936 18209 D ReactNativeBleManager: Read: 841F0009F5731AE709F086A2E86DF9775B46E5DF from peripheral: 80:EA:CA:70:01:13
02-25 03:23:26.622 17936 18209 D ReactNativeBleManager: Read: 841F000A83A3B8B61A7BDBC6B5CE7D63390E3680 from peripheral: 80:EA:CA:70:01:13
02-25 03:23:26.624 17936 18209 D ReactNativeBleManager: Read: 841F000B9CF6F5B872A018BE488B5F6725A8705E from peripheral: 80:EA:CA:70:01:13
02-25 03:23:26.626 17936 17949 D ReactNativeBleManager: Read: 841F000C785A3E8D60E79DC8705CB7D2238D61D7 from peripheral: 80:EA:CA:70:01:13
02-25 03:23:26.628 17936 17949 D ReactNativeBleManager: Read: 841F000DC5C76C5CE7DA0C8D62580DE16C9AE75A from peripheral: 80:EA:CA:70:01:13
02-25 03:23:26.769 17936 17949 D ReactNativeBleManager: Read: 841F002A489A308B20B759B66D5DA812418EE357 from peripheral: 80:EA:CA:70:01:13
02-25 03:23:26.771 17936 17949 D ReactNativeBleManager: Read: 841F002BB9CA6BDB7786168C23F879DD6BDB2725 from peripheral: 80:EA:CA:70:01:13
02-25 03:23:26.774 17936 17949 D ReactNativeBleManager: Read: 841F002CF9886398D9FE771BB6E5EC83A398C603 from peripheral: 80:EA:CA:70:01:13
02-25 03:23:26.776 17936 18209 D ReactNativeBleManager: Read: 841F002D7A1BF701D17EA1C8E6227C9DF6C9BB7B from peripheral: 80:EA:CA:70:01:13
02-25 03:23:27.030 25410 25410 I crash_dump64: obtaining output fd from tombstoned, type: kDebuggerdTombstone
02-25 03:23:27.035 17936 17950 D ReactNativeBleManager: Read: 841F002E2158D22884DED6F9AD74DFE8F6428A9F from peripheral: 80:EA:CA:70:01:13
02-25 03:23:27.038 17936 17950 D ReactNativeBleManager: Read: 841F002FA715AF6C9D6856378E62478DBC6A1BC7 from peripheral: 80:EA:CA:70:01:13
02-25 03:23:27.040 25410 25410 I crash_dump64: performing dump of process 17936 (target tid = 17971)
02-25 03:23:27.052 17936 17950 D ReactNativeBleManager: Read: 841F0030DA218FE2782DD9709BF73DFC8C639841 from peripheral: 80:EA:CA:70:01:13
02-25 03:23:27.055 17936 17950 D ReactNativeBleManager: Read: 841F0031F2705B8705F186A3B8C1FE735BF731E7 from peripheral: 80:EA:CA:70:01:13
02-25 03:23:27.058 17936 17950 D ReactNativeBleManager: Read: 841F00328623B8BA1379DB4695C17CA2190A4482 from peripheral: 80:EA:CA:70:01:13
02-25 03:23:27.058 25410 25410 F DEBUG   : *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
02-25 03:23:27.059 25410 25410 F DEBUG   : Build fingerprint: 'samsung/gts4lvwifixx/gts4lvwifi:9/PPR1.180610.011/T720XXS1ASL3:user/release-keys'
02-25 03:23:27.059 25410 25410 F DEBUG   : Revision: '5'
02-25 03:23:27.059 25410 25410 F DEBUG   : ABI: 'arm64'
02-25 03:23:27.059 25410 25410 F DEBUG   : pid: 17936, tid: 17971, name: mqt_js  >>> com.auraer <<<
02-25 03:23:27.059 25410 25410 F DEBUG   : signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x0
02-25 03:23:27.059 25410 25410 F DEBUG   : Cause: null pointer dereference
[... 93 lines cut ...]
02-25 03:23:27.472 25410 25410 F DEBUG   :     #69 pc 000000000027a854  /system/lib64/libart.so (bool art::interpreter::DoCall<false, false>(art::ArtMethod*, art::Thread*, art::ShadowFrame&, art::Instruction const*, unsigned short, art::JValue*)+940)
02-25 03:23:27.472 25410 25410 F DEBUG   :     #70 pc 0000000000529488  /system/lib64/libart.so (MterpInvokeStatic+204)
02-25 03:23:27.473 25410 25410 F DEBUG   :     #71 pc 000000000054b794  /system/lib64/libart.so (ExecuteMterpImpl+14612)
02-25 03:23:27.473 25410 25410 F DEBUG   :     #72 pc 0000000000363546  /dev/ashmem/dalvik-classes.dex extracted in memory from /data/app/com.auraer-E-uMsWjy_IXosEUcnnzGAw==/base.apk_17936_17936 (deleted) (com.facebook.react.bridge.queue.MessageQueueThreadImpl$4.run+74)
02-25 03:23:27.473 25410 25410 F DEBUG   :     #73 pc 0000000000254574  /system/lib64/libart.so (_ZN3art11interpreterL7ExecuteEPNS_6ThreadERKNS_20CodeItemDataAccessorERNS_11ShadowFrameENS_6JValueEb.llvm.3665920253+488)
02-25 03:23:27.473 25410 25410 F DEBUG   :     #74 pc 000000000025a068  /system/lib64/libart.so (art::interpreter::ArtInterpreterToInterpreterBridge(art::Thread*, art::CodeItemDataAccessor const&, art::ShadowFrame*, art::JValue*)+216)
02-25 03:23:27.473 25410 25410 F DEBUG   :     #75 pc 000000000027a854  /system/lib64/libart.so (bool art::interpreter::DoCall<false, false>(art::ArtMethod*, art::Thread*, art::ShadowFrame&, art::Instruction const*, unsi [... cut ...]
02-25 03:23:27.475 17936 17950 D ReactNativeBleManager: Read: 841F00B48224390A267C9C56B9AF7921D8A23B85 from peripheral: 80:EA:CA:70:01:13
02-25 03:23:27.479 17936 25413 D ReactNativeBleManager: Read: 841F00B55D66F1B67520A8B63087DF5729BE731E from peripheral: 80:EA:CA:70:01:13
02-25 03:23:27.484 17936 17950 D ReactNativeBleManager: Read: 841F00B6C87E2E8A615769C66F9D0836298DE247 from peripheral: 80:EA:CA:70:01:13
02-25 03:23:27.490 17936 17950 D ReactNativeBleManager: Read: 841F00B791C56C5BD7C61A8DA2F831D56D1AC709 from peripheral: 80:EA:CA:70:01:13
```

marcosinigaglia, 2020-02-28:
Hi @anwarhamr , I think your situation is particular since the app is active for so long. Are you reading the value from the device or are you using the notification? I'm not an Android expert so I don't see any hint in the exception.

You can create a new method to cache in memory or in filesystem the data you read for an interval and then transfer with a single call to the RN app so you can limit the bridge use.

anwarhamr, 2020-03-01:
Thanks for the reply.  My solution is for collecting data from 4 medical devices via characteristic notifications.  Thanks for looking at the errors.  I do think it is something w/ the amount of data that is being pushed so often... I'll play w/ making some changes and see what you think.  Thanks.

anwarhamr, 2020-03-07:
I'm trying to get started w/ the android side of the changes.  Do you have a contribution page w/ instructions and setup or could offer some guidance.  
I am struggling w/ getting the cloned project to compile.  I have the latest Android studio when syncing gradle I got `Could not find method google() for arguments [] on repository container` 
Searching online I found the recommendation to upgrade to gradle 4.1 [here](https://stackoverflow.com/questions/47152282/gradle-could-not-find-method-google-for-arguments-on-repository-container) 
and now I'm getting `cannot find symbo import androidx.annotation.Nullable;` 

 Thank you for your patience.

anwarhamr, 2020-03-08:
So I got it to compile but I'm def. using the wrong version of something.  It's telling me that `onNewIntent` is not an overwritable function in base ReactContextBaseJavaModule and `onActivityResult` has a different signature.  Cheers

marcosinigaglia, 2020-03-09:
Probably you are importing a wrong version of RN in the project,  for the androidx problem you have to use a version of the build tools that support AndroidX (https://developer.android.com/studio/releases/build-tools)

anwarhamr, 2020-03-09:
OK I'll look at that.  I have android buffering and it's working well for my solution; I also think it would be work contributing.  I do not have time to do the IOS part...   If you'd like to see it I can make a PR in the next couple days.  Do you have a linting file that you like, I'm using one and the modified files look more modified than they really are due to my settings.  VS code is helpful but sometimes nsm when you have prettier enabled. :/

marcosinigaglia, 2020-03-10:
If the caching part is separated go with the PR.

anwarhamr, 2020-10-03:
Buffer added to android

closed 2020-10-03 by anwarhamr