innoveit/react-native-ble-manager #597: Can BleManagerDidUpdateValueForCharacteristic receive more than 24 characters on Android?

hienpham2102, 2020-03-13:
### Version

Tell us which versions you are using:

- react-native-ble-manager ^7.1.4
- react-native 0.61.5
- Android 9.

### Expected behaviour: BleManagerDidUpdateValueForCharacteristic receives more than 24 characters.

I am using notification of ble  to receive 1 string  from a hardware via bluetooth connection... The library worked well on IOS but  for android , it got a string which only had 24 characters! Now, I want it to get more than 24? ( maybe 263 characters). Anyone can help me?

hienpham2102, 2020-03-16:
Hi, I find out  the reason,... I must call the function of requestMTU to expand MTU... This is resolved

closed 2020-03-16 by hienpham2102

preflower, 2020-06-04:
hi, i have same problem; and i call the function of requestMTU after connect before startNotification, but received data still 20 byte. can you tell me your call steps?