innoveit/react-native-ble-manager #602: Unable to find mobile devices near me. It fetches a bluetooth earphone and a laptop's data only.

ChiragDugar, 2020-03-31:
### Version

Tell us which versions you are using:

- react-native-ble-manager v?.?.?
- react-native v?.?
- iOS/Android v.?

### Expected behaviour

Tell us what should happen

### Actual behaviour

Tell us what happens instead

### Steps to reproduce

1.
2.
3.

### Stack trace and console log

Hint: it would help a lot if you enable the debugger ("Pause on exceptions" in the "Source" panel of Chrome dev tools) and spot the place where the error is thrown

```
```

Utkarshkr1710, 2020-04-01:
I am also getting the same problem..it is scanning only bluetooth earphones, band and laptop but not phone bluetooth

marcosinigaglia, 2020-04-01:
https://github.com/innoveit/react-native-ble-manager/issues/601

closed 2020-04-01 by marcosinigaglia