innoveit/react-native-ble-manager #603: getDiscoveredPeripherals method is not showing HC-06 bluetooth module devices

hakanpinar, 2020-04-01:
### Version

Tell us which versions you are using:

- react-native-ble-manager v7.2.0
- react-native v0.62.0
- iOS 13 / Android 9 

### Expected behaviour

Showing a list of connectable bluetooth devices included hc-06 module bluetooth devices

### Actual behaviour

getDiscoveredPeripherals method is not showing hc-06 module bluetooth devices

### Steps to reproduce

1. Start 
2. Start scanning
3. Showing a list in console
4. Trying to connect with the peripherals id

Code: 
`  useEffect(() => {
    BleManager.start({showAlert: false});
    console.log('useEffect');

    NativeAppEventEmitter.addListener('BleManagerStopScan', () => {
      // Scanning is stopped
      setIsScanning(false);
      console.log('stop');
    });

    NativeAppEventEmitter.addListener(
      'BleManagerDiscoverPeripheral',
      _getDiscoverDevices,
    );

    NativeAppEventEmitter.addListener('BleManagerDidUpdateState', args => {
      console.log('args', args);
      // The new state: args.state
    });
  }, []);

  function _startScanning() {
    setIsScanning(true);
    BleManager.scan([], 5, true).then(() => {
      _getDiscoverDevices(null);
    });
  }

  function _getDiscoverDevices(devices) {
    BleManager.getDiscoveredPeripherals([]).then(peripheralsArray => {
      // Success code
      console.log(JSON.stringify(peripheralsArray));
    });
    console.log('devices');

    BleManager.getBondedPeripherals([]).then(bondedPeripheralsArray => {
      // Each peripheral in returned array will have id and name properties
      console.log(JSON.stringify(bondedPeripheralsArray));
    });

    BleManager.getConnectedPeripherals([]).then(peripheralsArray => {
      // Success code
      console.log('Connected peripherals: ' + peripheralsArray);
    });
    console.log(devices);
  }

  function _connect() {
    console.log('testt connect');
    BleManager.connect('69:19:34:0F:A6:BF')
      .then(() => {
        // Success code
        console.log('Connected');
      })
      .catch(error => {
        // Failure code
        console.log(error);
      });
  }`

Is it normal that hc-06 mobule bluetooth devices not showing?

Thanks!

marcosinigaglia, 2020-04-03:
Try with an app like https://apps.apple.com/us/app/lightblue/id557428110

marcosinigaglia, 2021-10-11:
@gfmoore , yes, you need a Bluetooth 4.0 device that support BLE.

closed 2021-10-11 by marcosinigaglia