innoveit/react-native-ble-manager #604: writing not permitted

mahpgnaohhnim, 2020-04-02:
### Version

Tell us which versions you are using:

- react-native-ble-manager "^7.2.0",
- react-native "^0.61.5"
- iOS 12.4.5

### Expected behaviour

sending write should work on peripheral device with IOS
you can see the device here:
https://fccid.io/2ANKDJCP35NBLT

### Actual behaviour

BleManager.write(deviceId, serviceUUID, characteristicUUID, command, command.length).then(() => {
            console.log('Sent ' + commandName + ' command');
        }).catch(err => {
            console.log('write err', err);
        });

This actual works on Android, but on iOS i always get following error:
Writing is not permitted.

Even though i am using the same device, the same serviceUUID and the same characteristicUUID.

### Stack trace and console log

```
-----------------------------
Message to write(6): (
241,
241,
7,
0,
7,
126
)
-----------------------------
Message to write(6): f1f10700077e
-----------------------------
write err Writing is not permitted.
```

mahpgnaohhnim, 2020-04-02:
the characteristicUUID has the property "WriteWithoutResponse".
if i try another one for example ["Read","Write"] another Error Code will be shown:
```
Error Domain = CBATTErrorDomain Code=13 "The value's length is invalid." 
UserInfo={NSLocalizedDescription=The value's length is invalid.}
```

mahpgnaohhnim, 2020-04-02:
oh i am so dumb...
it should be `BleManager.writeWithoutResponse` instead of `BleManager.write`

closed 2020-04-02 by mahpgnaohhnim

AnshuRao, 2023-09-22:
@mahpgnaohhnim How to find  serviceUUID, characteristicUUID, ?